ArduPilot/ardupilot #5771: Ability to fetch DF name over telemetry

WickedShell, 2017-02-22:
#### Issue details
One of the most consistent problems when asking a user for the tlog/df log from their flight is that they provide you with a different tlog then the df log. This makes it extremely hard to tie them together. Because of timestamping issues (especially if they are missing a GPS on the aircraft) it can be hard to select the correct log.

If we could fetch the name of the current DF log over MAVLink while we are writing to it, then we can change the standard naming of tlogs to be {DF_LOG_NAME}-{TIMESTAMP}.tlog rather then the current {TIMESTAMP}.tlog standard that leads to a lot of confusion. To the best of my knowledge there isn't a way to poll for the current df log name in real time. (If there is then we just need to start asking GCS's to support this naming convention).

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

peterbarker, 2017-02-23:
On Wed, 22 Feb 2017, WickedShell wrote:

> If we could fetch the name of the current DF log over MAVLink while we are
> writing to it, then we can change the standard naming of tlogs to be
> {DF_LOG_NAME}-{TIMESTAMP}.tlog rather then the current [...]

I think a "current session UUID" would be a better option here.  This 
would periodically be logged in the dataflash logs, and periodically sent 
out as a mavlink message.

"current session" could be an AutoPilot thing, not a logging thing.  If a 
UAV-session-per-dataflash-log is desired then becase DataFlash is 
responsible for log opening/closing (LOG_FILE_DSRMROT) some sort of hook 
would be required to change the session ID.

This has the advantage that you can freely rename files (e.g. by 
downloading them using the mavlink log-download feature), and still be 
able to correlate df logs with mavlink logs.

WickedShell, 2017-02-23:
If the UUID is fundamentally in the log name then sure, however the use case I'm worried about is the SD card was pulled, how does the user know which log is associated with which tlog unless they can compare the names? (Needs to not require any external tool).

I'm fine with the UUID in the logs as well, but we need to start with the logs matching the tlog names, that makes the user experience a lot better.

peterbarker, 2017-02-23:
On Wed, 22 Feb 2017, WickedShell wrote:

> If the UUID is fundamentally in the log name then sure, however the use case

It isn't, and given filesystem limitations, can't be, I'd guess.

> I'm worried about is the SD card was pulled, how does the user know which
> log is associated with which tlog unless they can compare the names? (Needs
> to not require any external tool).

Ah, I was assuming external tool.  I run one to pull data off my SD cards 
and correlate them with my flight-logs directory tlogs (it moves the .bin 
files into those directories).

> I'm fine with the UUID in the logs as well, but we need to start with the
> logs matching the tlog names, that makes the user experience a lot better.

There are more issues here.  Typically a tlog will be created before a 
dataflash log is.  There may be several dataflash logs corresponding to a 
single tlog (and vice-versa).

I think it has to be content-based, sorry :-)

WickedShell, 2017-02-23:
Not at all! The GCS isn't subject to the same limitations and can rename the tlog file as needed. It isn't an RTOS and isn't stuck with a poor FS implementation. Even the multiple DF -> tlog isn't that bad as you can concatenate the names on the TLOG everytime you get a new one (or express a range {DF_LOG_NAME_1}-{DF-LOG-NAME_N}-{TIMESTAMP}.tlog.

The real problem with content based is thats not something users are doing when they fetch logs for reporting errors. Asking them to install *another* tool is problematic, and the scan time to resolve which UUID is correct is just far to expensive. You can't be asking this tool to scan a 3GB tlog folder and a 2GB tlog folder and do it quickly for content based. But the names can easily be managed.

peterbarker, 2017-02-27:
@WickedShell actually, depending on the OS the GCS might have significant issues renaming the tlog :-)

On that UUID concept:
https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L3871

I choose to ignore the "should correspond to the name of logfiles" bit, 'though ;-)

The intent of that message is to have a UUID-per-flight ("arming").  That has to come from the content of a dataflash log, not the name, as the user may not have chosen to rotate-on-disarm.  A ground-based GCS could choose to rotate to another tlog based on that mavlink message, of course.  I really think having the GCS be able to correlate the logs based on such a UUID is a good way forward.

rmackay9, 2017-02-27:
I'd like to add that apsync potentially resolves this problem (for people who use apsync) because the vehicle side can store both tlogs and df logs together.  Most people don't use apsync yet but I wanted to mention this anyway.

WickedShell, 2017-02-27:
@peterbarker What OS would have the significant problem? Worse case is you close, move, open the file again.

Rotate on disarm would mean a new log file anyways so I don't see the issue? Worst case you could split the UUID field so that the lower half is the log name, the upper half counts arming cycles.

I'm very infavour of ways to correlate logs, I just don't think opening and scanning for content is an acceptable solution on 3GB+ of bin files. The turnover time for that is just much to high. I mean a GCS could even be internally maintaining the list of tlog->df log pairing information if it can receive over the air the name of the current target log. That seems to me to be very straightforward information to expose to the GCS so that the GCS can take intelligent action after that.

jaxxzer, 2017-04-27:
What I've done in the past is send a SYSTEM_TIME message from the GCS to the autopilot, if the autopilot hasn't had a GPS fix, it sets the system time according to the GCS time. I think it's a super simple way to help with this problem in most cases.