ArduPilot/ardupilot #5772: Rover: Solo Cube lights go out

gmorph, 2017-02-23:
#### Issue details

http://discuss.ardupilot.org/t/how-do-i-install-apmrover-2-in-pixhawk-2/14394/6

A user (and I've replicated) pulled a Pixhawk Solo Cube out of a Solo and tried to install Rover on it.  The lights don't work once ArduPilot has finished booting.

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover

#### Hardware type
Solo Pixhawk Cube

peterbarker, 2017-02-23:
On Wed, 22 Feb 2017, Grant Morphett wrote:

> A user (and I've replicated) pulled a Pixhawk Solo Cube out of a Solo and
> tried to install Rover on it. The lights don't work once ArduPilot has
> finished booting.

That's be design.  The interfere with the compasses.

gmorph, 2017-02-23:
Really - they stay'ed on in Plane.  I didn't test Copter.

gmorph, 2017-02-23:
Should I raise it as a plane issue?

gmorph, 2017-02-24:
I've raised it as a plane issue.

proficnc, 2017-02-24:
This is 100% by design.  This issue should be closed

gmorph, 2017-02-24:
Close the issue.  Still not sure why its ok for the lights to go out on Rover but stay on in Plane however clever people then I say its ok so closing.

closed 2017-02-24 by gmorph

peterbarker, 2017-02-26:
On Fri, 24 Feb 2017, Grant Morphett wrote:

> Close the issue. Still not sure why its ok for the lights to go out on Rover
> but stay on in Plane however clever people then I say its ok so closing.

The lights go out on Plane when I load that firmware onto a PixHawk2 here.