ArduPilot/ardupilot #5775: Transition to more double-precision maths

magicrub, 2017-02-24:
In an effort to take advantage of future hardware that offer hardware assisted double-precision computation and the adoption of RTK, the codebase should be improved to use 64bit instead of 32bit precision where appropriate:

- gnc/EKF maths
- GPS RTK drivers
- GPS related maths (maybe already done?)
- mission waypoint support (new mavlink v2 packets?)
- position maths for various helper functions

There are a few options to consider, dependent on the build target:
- use "# if USING_DOUBLE_PRECISION" to declare double instead of float variables
- duplicate functions with float->double or int32_t->int64_t

Here are build targets to consider:
- Anything POSIX (x86/ARM: RasPi, QuRT, x86, ...)
- FMUv7

Nishchay-sopho, 2017-04-07:
Can we simply change the variables to double (or as prescibed USING_DOUBLE_PRECISION) in the header files and the cpp files to solve this issue?

amilcarlucas, 2021-09-22:
Some of this has been recently implemented by tridge.