ArduPilot/ardupilot #5778: Problem with fixed wing

juvinski, 2017-02-24:
#### Issue details
Hi All,
I made a RVJet Wing and I was flying pretty well, and using auto takeoff and auto landing.
I made several flights last wednesday and yesterday I was trying to flight with same setup, configurations and flight plan, but the takeoff was not working. 
Analyzing logs I couldn't find the reason the plane was not taking off - No error at least in my analysis. I download de wing configuration, put inside the simulador and the problem of not taking off was occuring too, with Bad AHRS error.
What is concerning my is. First I armed the wing successfully, manual/fwba mode was ok, no error reported. I sent the mission, armed and change to auto mode, when lauching my plane(with a bungee launcher) the place didn't start the motors. If is really a Bad AHRS problem, why before arm or change to auto mode there is any alarm or notification or prearm error to indicate there was a problem with the aircraft ? 

#### Version
ArduPlane-3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Fixed wing RVJet model

#### Hardware type
BBBMini (beaglebonoe + cape)

#### Logs
https://drive.google.com/drive/folders/0B1omx8hPUswTYTZRVjVXSlU1dU0?usp=sharing

juvinski, 2017-02-24:
HI,

in true I was using QgroundControl and now testing the simulation with Mission Planner, all the time Mission planner is showing the alarm, so I thing this is an issue with QGround not with ardupilot.

closed 2017-02-24 by juvinski

magicrub, 2017-02-25:
You must have the arm checks disabled. You should not be able to arm with
bad ahrs. That's the EKF taking too long to initialize due to GPS or IMU or
compass problems that it's trying to compensate for.

magicrub, 2017-02-26:
What is your arming_check Param value?

juvinski, 2017-02-27:
Hi Tom,

the arming check flag is set to all