ArduPilot/ardupilot #5780: Bit flags are flipped for Copter.

markrjr, 2017-02-27:
#### Issue details
I sent the command [SET ATTITUDE_TARGET](http://mavlink.org/messages/common#SET_ATTITUDE_TARGET) to a PixHawk running ArduCopter 3.4.5. For the type_mask field, if I don't set the bit, the fields are ignored. This is the opposite functionality of what the above page suggests. I was able to determine this by both testing on a real vehicle and by debugging with GDB. If you send the attitude command through the MavLink console, it has a bitmask of `63` which is 0b00111111 in binary and the motors actually spin up and the drone does the correct attitude. Though the functionality should be the opposite. For each set bit in that type_mask, those fields should be ignored.

#### Version
3.4.5

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad X.

#### Hardware type
PixHawk

#### Logs
_Please provide a link to any relevant logs that show the issue_
[image]

`STANDARD_ATTITUDE_BIT_FLAGS = 0b00000011`

khancyr, 2017-02-27:
Hi,

Sending bitmask of 63 which is 0b00111111 is the good thing ! As bit 7: throttle, bit 8: attitude are not set to 1 so they are not ignored and the copter follow the order, other bits are not implemented.
sending attitude message with STANDARD_ATTITUDE_BIT_FLAGS = 0b00000011 will pass too ... as only bits 7 and 8 are used... and you set bits 0 and 1  to 1 ...

OXINARF, 2017-02-27:
@markrjr I'm sorry I induced you in the wrong direction. I don't know how, but I ended up looking at the original implementation instead of what is now in master. The bug was actually fixed right after support for SET_ATTITUDE_TARGET was introduced to Copter: https://github.com/ArduPilot/ardupilot/commit/f3cbbef41800d820c96a0f7bdff33d53f9fc042f

Like @khancyr said, you are looking at the bits in the opposite way, you start counting from the right, not from the left.

closed 2017-02-27 by OXINARF