ArduPilot/ardupilot #5781: EKF2/3: assumes only 2 range finders

rmackay9, 2017-02-27:
There are likely issues with the EKF's use of the range finder class if more than 2 range finders are used:

- then "sensorIndex" is always 0 or 1 but the range-finder class can actually support up to 4 range finders (if the user compiles their own code and sets the (AP_RANGEFINDER MAX_INSTANCES to "3" or "4") meaning the EKF may not pull in values from the 3rd and 4th range finder.  Definition is here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.h#L24.
- In AP_NavEKF2_Measurements.cpp the "rngOnGnd" parameter comes from the primary range finder (or first downward facing range finder if Randy's rangefinder-orientation changes have gone in) but this value is used later against all range finders which could have different values for their, "_GNDCLEAR" parameter.

priseborough, 2017-02-27:
I need  to fundamentally re-think how we are combining range finder data. In the not too distant future, short range versions of these sensors will be available cheaply, enabling a median select voting algorithm to be used with three or more sensors for high reliability ground proximity detection, Doing consolidation in the  EKF does not scale well.

OXINARF, 2017-02-27:
I think @priseborough has a good point here. Why, for every orientation the rangefinder library can provide, don't we just return the best value we have? If there's only two rangefinders for one direction, I think we should return the shortest one, more than that and we can use the median algorithm Paul proposed.

rmackay9, 2017-02-28:
Yes, I agree .. the range-finder lib currently returns the values from the first one that matches the requested orientation but it could certainly do something better.  I don't think I've ever heard of a user attaching more than a single range-finder in the same orientation but I can easily imagine they will.

gmorph, 2017-02-28:
A Rover want's to use range-finder's differently.  A rover doesn't want to use them in the AHRS system but rather wants to use them in decision making i.e. stop or to go around the obstacle(s).  Copters are similar I guess from the obstacle avoidance point of view but range-finders have a dual use in Copter which is altitude information - the same way a plane uses it.  To this end the Rover code wants to know how far away an object is and in which direction probably in body frame.  My point here is I don't think we want too much logic built into the range-finder lib.  It should just say I have found an object this far away in this direction.  Note rover may want to know where each object is from each range-finder i.e. the object in front is 1m away but the object behind is 5m so we can back up a couple of meters and then go around.
Hmmmm, putting a lot of higher decision logic into the ArduPilot code.  As long as its contained in a library I guess that's ok.

Thanks, Grant.