ArduPilot/ardupilot #5784: [enhancement] - Improved Mixers for ardupilot with mavlink support

crashmatt, 2017-02-28:
#### Issue details
Need to support mixing for a wider range of vehicles and support real time tuning of those mixers.

#### Version
Future

#### Platform
[ X] All
[ ] AntennaTracker
[ X] Copter
[ X] Plane
[ X] Rover

#### Airframe type
This is mostly a benefit to fixed wing should be good for others also.  eg. multicopter users will be able to edit motor mixers directly.

#### Hardware type
Target all hardware

**SUMMARY**
There are two parts to this.  
1. Mixer tuning through mavlink
2. Creating a new mixer architecture capable of supporting all vehicles 

2 should come before 1 but due to forces of nature 1 has arrived before 2.  The mavlink support has been prototyped into mavproxy and is being prototyped into qgroundcontrol.

**Read before you panik** The mavlink mixer service is planned to be as flexible as possible.  It attempts not to constrain the mixer architecture at the vehicle.  Using this service means you get gui support for free.  

Here are some support documents:

[MAVlink mixers tuning document
](https://docs.google.com/document/d/1LeQC2qtxqdesNuJ3RAWH_MJ_30xjf-PMN3bd5wtUAUo/edit?usp=sharing)

[Improved mixers document - Including requirements and proposals
](https://docs.google.com/document/d/1QAnT1GI0ejSPHQZfFSnYyyOMAejQkxnzNF6W1-p1Pk8/edit#heading=h.tsjtmv2wzz6k)

**ACTIONS**

closed 2017-03-25 by crashmatt

magicrub, 2017-03-25:
?

crashmatt, 2017-03-25:
Closing due to no active discussion.

magicrub, 2017-03-26:
So... Why close it?

crashmatt, 2017-03-26:
I am already maintaining too many discussion threads and this one is dead.  The ardupilot development forum would be ideal but I can't post there.  May re-open once I am ready and have new documentation.