ArduPilot/ardupilot #5785: ll40ls: inmediate read failed: try again.

marv3, 2017-02-28:
Hey, I'm using a Lidar Lite v3 but when I try to test this(ll40ls test i2c), nsh displays:

ll40ls: inmediate read failed: try again.

I tried a lot of times and it doesn't work.
Can you help me? Please.

(I' using a hexacopter with a pixhawk)
THANKS.

#UPDATE 1: When I try using "ll40ls info i2c" there is a distance parameter which displays the measured distance (I tried in two different positions). But, this "distance" doesn't change if I change the position of the lidar (to measure other distance), when I try with "ll40ls info i2c" the display for "distance" is the same.
HELP! and thanks again.
[image]
[image]

OXINARF, 2017-02-28:
Several points:

- user issues should be posted at the forum (http://discuss.ardupilot.org) and not here
- we don't use that PX4 driver anymore, we use our own drivers now
- Lidar Lite v3 isn't supported yet

closed 2017-02-28 by OXINARF

marv3, 2017-02-28:
Thanks for your answer and sorry for posting here (first time mistake).
Can you answer just a last question, please? I really need to know how to work.
I tried with the beta version released 7 days ago and it doesn't work (thanks for the third point).
Is there a guide to develop drivers or something like that? I really need to work with the Lidar Lite v3.

Thank you and sorry again.

OXINARF, 2017-03-01:
If you want to compile it yourself, supporting **just** v3 is fairly easy. You need to remove the [hardware/software version check](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_PulsedLightLRF.cpp#L153-L161) and then make [`v2_hardware == true`](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_PulsedLightLRF.cpp#L164).