ArduPilot/ardupilot #5787: Plane: battery failsafe alarm when plug the main power

robustini, 2017-02-28:
I noticed that was removed this feature, which alerts with beeps and warning the planner when connecting a main power battery with lower voltage than the failsafe setting.
If possible it would be better to restore this thing.

magicrub, 2017-02-28:
Hi @robustini , does pending PR (https://github.com/ArduPilot/ardupilot/pull/5680) fix this? That PR is Plane only but the same problem may apply to copters too.

robustini, 2017-02-28:
No, yestarday i crash a VTOL during takeoff due battery failsafe.
I know there's new parameter that prevents to arm the system when the battery has too low, but it would be preferable to have this indication before arming, the buzzer helps a lot.

robustini, 2017-02-28:
With the copter (battery failsafe set to "21V")...

[image]

robustini, 2017-09-04:
This issue is solved with another parameter, added in the latest release of the plane.

closed 2017-09-04 by robustini