ArduPilot/ardupilot #5788: Add a function which controls rc inputs after rc fail safe

githubtd01, 2017-02-28:
Add a function which only allows new rc intputs after rc fail safe and rc recovers when vehicle has arrived in a safe distance to home. Or requiriere to switch modes from RTH to other mode to reenable rc inputs.
requires.

####
Background:
Had the following issue: [http://discuss.ardupilot.org/t/copter-switches-from-auto-to-stabilze-and-back-after-rc-failsafe/15263](url), possibly connected to repeated loss and regain of rc signals?

OXINARF, 2017-03-01:
Regarding your issue, I'm not sure what it is. One person suggested you have your radio failsafe wrongly configured, another suggested that turning off your radio and then on isn't a good idea because it might initialize and send wrong values. Neither of these are an ArduPilot problem.

Your request seems to be a workaround for a bad configuration or a bad practice. If you lost RC input and then it recovered, why wouldn't you want immediate control?

@rmackay9 what do you think of this?

githubtd01, 2017-03-01:
My radio is correctly configured. Failsafe and recovery worked fine for at least 25 times. I never switched of the transmitter when the quad was in the air (only to test failsafe when installing the receiver). The described error occured only once and my idea is that this could be caused by constant rc loss and recovery?

My previuos transmitter required switching modes to regain control. I think that is quite a good thing because I think the most errors which occure during rc recovery will be due to false control of the radio. So this adds a layer of security, you check whether the transmitter controls are ok and then you reconnect by switching the modes.

OXINARF, 2017-03-02:
I gave detailed answer about your problem on your thread.

I understand your proposal, I still think it just complicates setup and I can see someone set it up and, in the middle of the stress of trying to regain control, forgetting what sequence is needed to regain control. I think ArduPilot has always tried that the pilot is at control at all times - even when in Auto mode you can **always** regain control. Having a combo to regain control after a radio failsafe goes a bit against that.

githubtd01, 2017-03-03:
Yes, excuse me, I'm now realising that I missunderstood the whole radio failsafe thing. I think this is obsolete. Excuse me.

OXINARF, 2017-03-03:
Ok, closed.

closed 2017-03-03 by OXINARF