ArduPilot/ardupilot #5789: Copter: advanced tuning like in the VTOL code

robustini, 2017-02-28:
I love the tuning system that @tridge has implemented for "QuadPlane" (VTOL), I hope @rmackay9  will consider this awesome implementation also the Copter:

http://ardupilot.org/plane/docs/common-transmitter-tuning.html

peterbarker, 2017-03-01:
This tree was the basis of a PR to get the same functionality in Copter IIRC: https://github.com/tridge/ardupilot/tree/pr-copter-tuning

It doesn't rebase cleanly.

OXINARF, 2017-03-01:
The PR for this is open, but hasn't been merged yet: #4065. Actually Randy mentioned @robustini there, (but without tagging) asking for your opinion.

Sometimes we leave issues and PRs for the same subject open, but that normally happens when PRs are opened to fix an issue. Since this was already at PR stage I'm going to close it. @robustini let me know if you think this should stay open anyway.

closed 2017-03-01 by OXINARF