ArduPilot/ardupilot #5795: Enhanced camera feedback configuration

githubtd01, 2017-03-01:
The  CAM_FEEDBACK logging is a great tool. But since I'm using simple but very light Canon Powershot A490/A495 I think I will not be able to use it. The only way to sync would be the LED which cannot be controlled via the KAP script but seems to blink once when the shutter is released and then again when the images are written to the sd card.

In order to use the led in combination with a (photo-)transistor the following logging behaviour/parameters would be needed:

1. only log a  CAM_FEEDBACK message when a trigger event occured
2. only record the first  CAM_FEEDBACK message/pin activity when a trigger event occured
3. ignore the following  trigger event occured messages/pin activities for x miliseconds

What I miss (I hope that is not due to ignorance) are some parameters to fine tune the  CAM_FEEDBACK logging.

1. a parameter to switch the behaviour descriped in nr. 1 on and off
2. a parameter to choose which (first, second,...) CAM_FEEDBACK message/pin activity should be logged (the behaviour descriped in nr. 2)
3. a parameter to define the number of miliseconds following messages/pin activities should be ignored, by standard the same value as CAM_MIN_INTERVAL. (the behaviour descriped in nr. 3).

ptsneves, 2017-03-15:
Can you confirm that when you have the feedback pin enabled you get a cam_feedback only when there is a trigger received from the camera?

WickedShell, 2017-03-15:
@ptsneves you get a CAM message anytime you detect the specified edge transition from the feedback after it has been enabled the first time.

ptsneves, 2017-03-15:
@WickedShell This was what I was thinking. This is why i am not understanding this issue. On a slightly offtopic, did you review the math of camera offsets?

githubtd01, 2017-03-16:
@WickedShell yes, exactly, a single CAM message is the thing I'm looking for.
@ptsneves & @WickedShell can I help by providing additional information like video of the led behavior?

This are the cameras:
http://chdk.wikia.com/wiki/A490 (the green state LED is blinking as described above, the AF LED is not working, I think that is due to slight differences in the code of CHDK, I can identify them but unfortunately I'm not capable to modify them)
http://chdk.wikia.com/wiki/A495 (additional to the green state LED the AF LED is blinking when the shutter is released but unfortunately also when the image has been written on the SD card.)

jmachuca77, 2017-03-16:
I don't think this is worth pursuing. The whole point of using the feedback pin is to have a precise trigger position for the camera. Since the LED on the camera is not exactly in sync with the shutter curtain. It would still introduce a timing error which would be about the same as not using the feedback pin and just logging the cam message the standard way which would be when the camera is sent the trigger command.  I would recommend to close this issue.

githubtd01, 2017-03-16:
ok, that is a valuable information, does that only apply for my cameras or also for a Canon S100? That would mean that this product is nonsense? 
http://www.tuffwing.com/support/Install_a_canon_precision_geotag_cable.html

jmachuca77, 2017-03-16:
Yeah, I don't think that cable would work, the red eye reduction led flashes a few seconds before the picture is taken, and then again when the shutter opens, this is to make the iris on the subject eyes close before the picture is actually taken so that there is no light being reflected back to the camera from the back of the eyes. So you will still get multiple feedback from that LED and it will not be in sync with the shutter either. Currently the only way to get an accurate sync is if the camera has a sync output, externa flash connector, or a hotshoe for a flash.

WickedShell, 2017-03-16:
I do agree with @jmachuca77. I didn't close it as I could see someone else implementing it, but I can't see going out of my way to implement it due to the lack of precision it would grant, the only thing it approaches is some mechanism to allow the GCS to know if it is probably taking photos...

githubtd01, 2017-03-16:
@jmachuca77 no, you misunterstood the concept. The CHDK script "KAP script" blinks the LED in the same (what ever that means) moment the shutter curtain is moved by the camera. It is far more precise than what you propose.

"optional use of a camera LED that illuminates precisely when the shutter opens for sync with an external device like a flight controller"

http://chdk.wikia.com/wiki/KAP_UAV_Exposure_Control_Script

jmachuca77, 2017-03-16:
@githubtd01 the links you posted before did not talk about direct control of an LED. If the KAP script is actually doing what it says then the sync would be better and the cable posted above would work. But then I don't see how this would cause multiple CAM messages. Do you have a scope trace of the feedback pin when using the phototransistor?

OXINARF, 2017-03-17:
We don't close feature requests just because the team isn't implementing them - if we did, we would have a much much smaller number of open issues. We should only close this if a PR with it would be rejected - and a decision like that would probably be taken at a dev call.

ptsneves, 2017-03-17:
@githubtd01 Second the request for the scope trace.

githubtd01, 2017-03-18:
@ptsneves excuse me, I do not know hat a scope trace is. Can you specify that please?
Is that inside the log?

jmachuca77, 2017-03-18:
An oscilloscope trace of what the feedback signal coming into the pixhawk looks like. This is to see is there are multiple pulses, or if it's a noise issue or something else

Saludos / Best Regards

Jaime Machuca Mercado
CTO | Droidika | www.Droidika.com
Mail. Jaime@Droidika.com
Tel. +52 (33) 3777 5327
Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Droidika S.A. de C.V., deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo

githubtd01, 2017-03-20:
I do not have an oscilloscope, can I retrieve that from the pixhawk log 
file?

jmachuca77, 2018-08-16:
Is this issue still valid? If it is we would need a scope trace of the signals to see whats going on and propose a solution.

githubtd01, 2018-08-16:
Thx for picking it up again and sorry for me not. I'm using now an Arduino to filter the signal. If there no one else interested in such a solution I think I would propose to close the issue.

closed 2018-08-16 by githubtd01