ArduPilot/ardupilot #5796: How To Create New Variable on Pixhawk (dev beginner question) ?

ardianflyer, 2017-03-01:
**#### Issue details**
Hi All. I'm new on Pixhawk coding development but I have already about 6 month on APM2 coding. I run toolchain on windows 10. In APM code I can create new variable easily, but I can't do it on Pixhawk. 
For example in order to trace thortle value, I create new variable in AP_Motors_Class.h :
static float           _thrust_rear2;

then I call it from APMotorTri.cpp :
 _thrust_rear2 = _thrust_rear;

the while i compile the code (px4-V2), i get error like this 
c:/DATA/ARDIAN/PIXHAWK/V5/ardupilot/libraries/AP_Motors/AP_MotorsTri.cpp:259: undefined reference to `AP_Motors::_thrust_rear2'

what's wrong? I already search all over internet but can't find answer until now, please help. Sorry for this very newbie question. Thanks in advance.

**#### Version**
I download code from github about 4 days ago 

**#### Platform**
[  ] All
[  ] AntennaTracker
[ *] Copter
[  ] Plane
[  ] Rover

**#### Airframe type**
Tricopter

**#### Hardware type**
Pixhawk

**#### Logs**
N/A

khancyr, 2017-03-01:
Hello,

You need to look at static keyword on C++ (https://www.google.fr/search?q=c%2B%2B+static&oq=c%2B%2B+static&aqs=chrome..69i57j69i61l3j69i59l2.2520j0j7&client=ubuntu&sourceid=chrome&ie=UTF-8#q=c%2B%2B+static+undefined+reference+&*) ! 
You miss to initalize your _thrust_rear2

ardianflyer, 2017-03-01:
Dear khancyr, thank you for you fast response. I'm following up your suggestion now. May be need some time to get result because I'am new on pichawk coding.

OXINARF, 2017-03-01:
@ardianflyer If you don't know C++ I strongly suggest you try to learn at least the basics of it, otherwise you'll have a hard time working with ArduPilot.

Also, this list is for confirmed bugs and features requests. For developers questions please post at the forum (http://discuss.ardupilot.org) or ask in Gitter (http://gitter.im/ArduPilot/ardupilot).

closed 2017-03-01 by OXINARF

ardianflyer, 2017-03-02:
yes i think still need more strugling on C++, thank you for suggestion