ArduPilot/ardupilot #5799: AC3.5: Pixracer params can't be updated over wifi

rmackay9, 2017-03-02:
If using a Pixracer with the little wifi board, changes to parameters (i.e. ARMING_CHECK) always fail.  This was tested on master (pre AC3.5-rc2) using windows PC and mission planner.

rmackay9, 2017-03-02:
First reported by artemen on this thread: http://discuss.ardupilot.org/t/copter-3-5-rc1-released-for-beta-testing

WickedShell, 2017-03-02:
@rmackay9 This is the same underlying issue as #5505 I think. It's not something master plane suffers from. If you wait long enough you can download params quickly, but its still very hit or miss.

rmackay9, 2017-03-02:
@WickedShell, ok thanks.  If it's the same issue then the bit of new info I have is that the performance is worse between AC3.4.5 and AC3.5-rc1.

WickedShell, 2017-03-02:
I believe @DonLakeFlyer was saying it fell apart around the time of the frame change? I could be really off on that recollection however.

OXINARF, 2017-03-02:
As I've written in discuss, the wifi port is just like another serial port from an ArduPilot perspective. If this is an issue, is it happening with other devices? Here is a [link](http://discuss.ardupilot.org/t/esp8266-wifi-on-pixhawk/14681) for a long discussion where @peterbarker was testing new firmware for the Wifi board and got much better results.

> Updating to the most recent MAVESP8266 firmware has solved the vast majority of the lost packets on the downlink - I'm at 1.2% packet loss.
> 
> Looking at it, it makes sense; prior to this change a lot of the packets the vehicle sent would not have been recognized - ArduPilot support is actually a late-comer to the ESP8266 firmware. This is a drawback in having the ESP8266 interpreting the mavlink messages.
> 

I don't know why it has gone worse with 3.5/master but it's possible we are sending more messages and the Wifi board just can't keep up.

WickedShell, 2017-03-02:
@OXINARF Plane (at least as of when I commented on #5505) wasn't suffering any problem like that.

OXINARF, 2017-03-02:
This should be fairly easy to test:

- connect a radio to the first telem port in Pixracer, does it work?
- connect the wifi board to the same telem port, does it work?
- connect the radio to the wifi port, does it work?

Don't forget to change the baudrates accordingly. If only the Wifi board doesn't work, it is almost certain that the issue isn't with ArduPilot.

rmackay9, 2017-03-03:
A bit more information:

- the issue affects Plane (master) as well as Copter (and probably all vehicles)
- the GCS->vehicle link is likely the cause of the slow parameter downloads (and inability to set parameter values) with many messages from the GCS not successfully getting through to the autopilot.

lvale, 2017-03-03:
I've been tracking this with Don here https://github.com/mavlink/qgroundcontrol/issues/4643

peterbarker, 2017-03-04:
If you could try `026c65d3646b0f5671bf5c5a7d58c5d2cc82844c` then try `026c65d3646b0f5671bf5c5a7d58c5d2cc82844c^` to see if you see the same thing that would be great.

Note you'll want a git submodule update between as it is a submodule commit in question.

peterbarker, 2017-03-17:
This problem appears to be resolved.

closed 2017-03-17 by peterbarker