ArduPilot/ardupilot #5806: PH2.1 : # Compasses > 3 => calibration issue

Kiwa21, 2017-03-03:
#### Issue details
When using Pixhawk 2.1 with an external compass, the total number of compasses available is 4.
Current MAX in Arducopter is 3.

Calibration of external compass only works fine.
When external is unplugged, calibration of internal compasses works fine.
However, we could see the Compass 1has bad values whereas Compass 2 and 3 have good values.

Both at the same time = Calibration becomes a pain when Compass 2 (which was compass 1 when internal only) has bad values.

Then if we try to do calibration witch Compass 2 unchecked (because bad values) Compass 1 External and 3 Internal checked, the Pre-Arm check "Compass not calibrated" appears even though the calibration was successful.

So 2 issues to be solved : 
- It seems that not using Compass 2 is "breaking" the calibration validation process.
- Compass 4 (good offsets) can't be used to replace Compass 2 because current limit is set to only 3.

If I am doing something wrong here, please advice :)
Thanks,

`AP_Compass.h
/**
   maximum number of compass instances available on this platform. If more
   than 1 then redundant sensors may be available
 */
#define COMPASS_MAX_INSTANCES 3
#define COMPASS_MAX_BACKEND   3
`

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Hardware type
PIXHAWK 2.1 (ProfiCNC)

OXINARF, 2017-03-03:
@Kiwa21 

- user issues need to be posted in the forum (http://discuss.ardupilot.org) and not here; this list is for confirmed bugs and feature requests
- if you see three compasses with the Pixhawk 2.1 it means you are not using master or even 3.5, which is one more reason why you should have posted at the forum and not here
- if you have an issue (like you pre-arm compass not calibrated message) a log is highly recommended to be posted as otherwise it makes it much harder to diagnose the problem
- I don't see us increasing the default number of available compasses; from numbers based on my own perception, 95% of the time only one compass is used and 99.9% of time two are, at most, used, so increasing it has very few advantages; in any case the library has been coded so that anyone can increase it and build their own firmware - open source has that advantage
- choosing what compasses you want to have when there are more than 3 isn't easy to do, but choosing the order between the 3 available is something we have on our agenda for the future; we already prioritize the external ones over internals (if an external is worse than an internal then why use it?) and we also try to use the internal ones in the best order

closed 2017-03-03 by OXINARF

Kiwa21, 2017-03-03:
@OXINARF My bad for posting here if you think it doesn't belong to this category.
However, this is a repeatable bug not a user problem. I already got it on 3 differents PH2.1
I don't know why you say we can't see 3 compasses on master or 3.5 ? I am running 3.5rc1 and I can clearly see 3 internals.

Choosing the order would be good enough already to solve this kind of issues.

OXINARF, 2017-03-03:
> However, this is a repeatable bug not a user problem. I already got it on 3 differents PH2.1

With the same user... I'm not saying there can't be a bug, but it's not been confirmed. You need to post in the forum together with a log.

> I don't know why you say we can't see 3 compasses on master or 3.5 ? I am running 3.5rc1 and I can clearly see 3 internals.

Because we don't use one of the compasses in Pixhawk 2.1 in 3.5 or master. If you are seeing 3 internal compasses you don't have 3.5, that's for sure.

> Choosing the order would be good enough already to solve this kind of issues.

I'm not sure how it would fix your issue. What I said is in our agenda is to choose the order between the three available compasses, you won't be able to choose any compass that is currently not shown.