ArduPilot/ardupilot #5813: Copter: AC3.5-dev - EKF never resets to GPS position after glitch

rmackay9, 2017-03-04:
During testing of master I've found it's possible to make the EKF maintain an offset after a GPS glitch occurs by doing this in the simulator:

- map set showgpspos 1  (to make the raw gps position visible as the blue vehicle)
- param set SIM_GPS_GLITCH_Y 0
- loiter, arm throttle, rc 3 1800, rc 2 1400 (to get the vehicle gently flying forward)
- param set SIM_GPS_GLITCH_Y 0.002

In AC3.4 the blue vehicle would jump to the right and a few seconds later the red vehicle (the EKF estimate) would reset to the GPS position.  In AC3.5 however the reset never happens so the two vehicles appear to be flying next to each other.

This problem does not occur in AC3.4.5 (the official release) so this only affects developers and beta testers.

rmackay9, 2017-03-04:
Note: this only affects the horizontal reset, the vertical reset (if using EK2_ALT_SOURCE=2/GPS) works.

WickedShell, 2017-03-04:
Oh good, was getting really worried about that part actually! Thanks for
the update!

priseborough, 2017-03-04:
I found it - when the reset logic was reworked to include the range beacon, the unintended result was that a position reset would not occur if GPS velocity observations were still being accepted. This didn't show up with my tests using replay logs because realistic glitches and IMU errors always caused the GPS velocity to be rejected as well as the position. 

The intended result was that without GPS velocity rejection, a reset would occur after 5 seconds or 10 seconds if GPs velocity was being fused.

Fixed by https://github.com/ArduPilot/ardupilot/pull/5816

rmackay9, 2017-03-06:
This is fixed in master now.  Thanks @priseborough

closed 2017-03-06 by rmackay9