ArduPilot/ardupilot #5817: APMRover2

khancyr, 2017-03-04:
#### Issue details
Ouput can be more that 2000ms on skid steer config...
To reproduce on sitl: arm throttle, rc 1 2000, rc 3 2000
[image]

#### Version
master : https://github.com/ArduPilot/ardupilot/commit/2631bdbe64e32a9607bda642125a87f5979aafc6

Not present on release APMrover 3.1.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X] Rover

OXINARF, 2017-03-06:
@khancyr Can you improve the title of this issue? It's not descriptive at all.

khancyr, 2017-03-06:
@OXINARF yep sorry, forget to fill it correctly

khancyr, 2017-05-04:
solve with SRV_Channel: limit pwm_from_angle return to +/-high_out #6190

closed 2017-05-04 by khancyr