ArduPilot/ardupilot #5818: Pixracer Autopilot 3.4.5 Gyro problem

alfadroid, 2017-03-04:
#### Issue details
iv'e buy a new pixracer and load Ardupilot 3.45 firmware, but i've a problem.
I set up all the sensor, soft tuning ad seam go fine.
But when i switch on for the first time and arm it for the first time, the drone fly fine, but when a rearm for the 3-4 time Qgroundcontrol tell me Arm check: gyro still setting and the drone is not stable.
If i see the gyro are not set correctly on horizon, but when i disarm ti fix yourself.
Then if switch off and switch on again the drone go fine and very stable until to the 3-4 time that i arm and disarm again i have the same problem.
What is the problem?

#### Version
3.45

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
X4

#### Hardware type
pixracer

#### Logs
https://www.dropbox.com/s/tq3l1se26okj03n/log_10_2017-3-4-17-03-28.bin?dl=0
https://www.dropbox.com/s/rxr9ufruqmq2c05/log_11_2017-3-4-17-09-02.bin?dl=0

alfadroid, 2017-03-04:
Update:
I've test an other pixracer board and the problem show after 3-4 arm/disarm.

OXINARF, 2017-03-06:
This is a support question and those should be posted in the forum (http://discuss.ardupilot.org).

closed 2017-03-06 by OXINARF