ArduPilot/ardupilot #5821: Reporting Auto Stages

WickedShell, 2017-03-05:
#### Issue details
From the GCS side it would be incredibly valuable to know what stage the aircraft (while in auto) is in. For example if you are in auto on plane you could be in takeoff, normal, landing, landing abort states. This information would be incredibly helpful to show the user for diagnosing what the aircraft is doing, as well as changing visualization features. We send some statustexts at the moment, but that isn't a reliable comms method as you can miss the message, or if you connect to the aircraft after it's in that state it won't be clear which state it is in.

What I don't see is a good way for us to expose this at the moment. There doesn't appear to be any generic MAVLink packet that lets us communicate this information. The best idea I've had so far is to grab one of the error_counts fields from SYS_STATUS ( http://mavlink.org/messages/common#SYS_STATUS ). There are 4 of these fields that are for autopilot specific error reporting, so we can assign the stages to them as a meaning, and we don't currently use any of them. However this feels like a pretty hackish way to communicate the information to the GCS, so I'm open to alternate suggestions that everyone would feel happy with.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

magicrub, 2017-03-06:
How about a new packet and it is sent every 10 seconds and also immediately on every mode change. The bigger question is what should the packet be. I don't like hijacking an existing one. Maybe we can make heartbeat2 ?

OXINARF, 2017-03-06:
I don't like hijacking error fields to send normal information - specially in a common MAVLink message.
Since this is only to be used only in Auto modes, a new message sent periodically is probably fine - we just need to have an enum with the possibilities for all vehicles.

WickedShell, 2017-03-07:
Dev call conclusion was to propose extending the HEARTBEAT message. Will post a link to that once I've submitted it.

peterbarker, 2024-07-31:
I'm a touch worried about how this might tie our internal state to external things (e.g. GCS).

I've really wanted this for things like Copter's mode-guided submodes, so I think it's worth doing.

Note that some of the goals of this can be achieved using the `EXTENDED_SYS_STATE` message we *do* have support for,