ArduPilot/ardupilot #5824: unneeded servo initialization

night-ghost, 2017-03-06:
#### Issue details
SRV_Channel/SRV_Channel_aux.cpp line 150

if (function < SRV_Channel::k_nr_aux_servo_functions) {

should be

if (function < SRV_Channel::k_nr_aux_servo_functions && function!=SRV_Channel::k_none) {

because current version initializes pins as servo even if pin not used, preventing of sharing pins for different needs

#### Version
git

OXINARF, 2017-03-06:
Have you read the comment there?

> enable all channels that are set to a valid function. This includes k_none servos, which allows those to get their initial trim value on startup

I've added this to the dev call today just to make sure anyway.

night-ghost, 2017-03-06:
YES I read this comment, but.
In the current case servo pin initialized ALWAYS which prevents of using non-dedicated pins as servos - for example unused pins of input port, which is actual for small controllers.

The question is: what a reason to initialize unused pins? Old APM times with a lots of free pins?

This change tested and working, and I can't imagine any drawback.

tridge, 2017-03-07:
That initialisation is needed for people using the MAVLink DO_SET_SERVO and DO_REPEAT_SERVO to do things like control package release. Without this the servo won't get initialised to the TRIM value on boot.
I know it isn't a great design, but just removing that code would break existing functionality

night-ghost, 2017-03-07:
Ok, understand. But there should be a way to load trims *without* actually reprogramming the pin before usage. Eg. to change k_unused1 value to something like "this servo controlled from script". Yes this will require to change existing scripts but we already have a lot of things changed between versions.
BTW, how much controllers has dedicated servo pins now?

davidbuzz, 2017-03-08:
I'd also like to point out that there are some instances where doing a GPIO initilisation on a pin is undesirable.   for example, when a pin is used elsewhere for DMA-enabled IO, then it should *not* be initialised the "normal" way.    I suspect that's going to be a more rare case though.      Having the option to go either wat would be helpful in some cases.     eg. SERIALX_PROTOCOL parameters take -1 as a value that means "this serial port not used at all".

night-ghost, 2017-03-23:
Now I can prevent unneeded initialization by HAL parameters so if this issue not supported by anyone it can be closed.
[image]

tridge, 2017-05-01:
@night-ghost what are those HAL_* parameters? which ArduPilot version are you using? It isn't in master
The way we leave pins uninitialised in master is with BRD_PWM_COUNT, which allows you to mark any of the AUX pins for use as GPIO instead of PWM. That gives up to 6 GPIO pins.

night-ghost, 2017-05-02:
>what are those HAL_* parameters?

just a way to pass configuration data directly to HAL. What gives a lots of useful things, eg. Cleanflight/OpenPilot/Arducopter motor layout switching or I2C/UART selection for FlexiPort, which can't be done in master.

> which ArduPilot version are you using? It isn't in master

Of cause, this is my fork for RevoMini which never will be at master, as I understand it.

>The way we leave pins uninitialised in master is with BRD_PWM_COUNT

this way allows to use just unused output pins, not unused inputs

IamPete1, 2021-07-25:
fixed by https://github.com/ArduPilot/ardupilot/pull/18018

closed 2021-07-25 by IamPete1