ArduPilot/ardupilot #5829: SERIAL5 does not read

patrickpoirier51, 2017-03-07:
#### Issue details
Trying to connect serial rangefinder to FC on SERIAL5 using a BBBMINI linux FC.
The BBB has 2 serials ports /dev/ttyO4 -and- O5 these physical ports are logically linked to Px4 SERIAL devices using Uart assignment : Uart A is assigned to SERIAL0 , B-1, C-2, D-3,  E-4 , F-5.
 
Launching
 ./ardupilot -F /dev/ttyO4
  ---Should assign rangefinder connected to BBB physical ttyO4 to the SERIAL5 (following the F-5 logic)--  
Settings:
SERIAL5_PROTOCOL = 9 (Lidar)
SERIAL5_BAUD = 19 (19200 baud)
RNGFND_TYPE = 8 (LightWareSerial) 

Get Bad Lidar Health loading Mission Planner

==================
Launching
/ardupilot -E /dev/ttyO4

Settings: 
SERIAL4_PROTOCOL = 9 (Lidar)
SERIAL4_BAUD = 19 (19200 baud)
RNGFND_TYPE = 8 (LightWareSerial) 

Works OK  loading Mission Planner

**NOTE: Tried with MAVLink as well == Same result**

#### Version
_What version was the issue encountered with_
Tried 3.4 & 3.5

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_ N/A

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
BBBMINI

OXINARF, 2017-03-07:
@lucasdemarchi Any idea? I've gone through code and couldn't see anything wrong, but I should be missing something.

khancyr, 2017-03-07:
@patrickpoirier51 Did you try with a GPS to see if the port is really working ? (eliminate ardupilot handle of uart fault) And without ardupilot, by connecting an arduino and open the com ? (eliminate material fault)

patrickpoirier51, 2017-03-07:
@khancyr  As described I tried with 2 different devices:  rangefinder(type 9)  and  mavlink (type 1) , with same result SERIAL5 does not not read. 
Look like the Uart to Serial5 ("-F")  assignment is wrong or broken in HAL_LINUX.  
@OXINARF you confirm that SERIAL5 is working ok on a Pixhaw or Pixracer?

patrickpoirier51, 2017-03-07:
TESTED WITH SITL

 Failed SITL test with sim_vehicle.py -A "--uartF=uart:/dev/ttyUSB0" and PASSED with sim_vehicle.py -A "--uartE=uart:/dev/ttyUSB0"
This is in accordance with this text: Any of the 5 UARTs can be configured in this way, using uartA to uartE. http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#using-real-serial-devices

closed 2019-02-07 by patrickpoirier51