ArduPilot/ardupilot #5830: how to place pozyx  anchors  to use IndoorLoiter.ino

walkeryu123, 2017-03-07:
I want to use the IndoorLoiter.ino to make copter loiter. after reading the code i found anchors should place at the same height. and  I have changed anchors_x[NUM_ANCHORS], anchors_y[NUM_ANCHORS] and heights[NUM_ANCHORS] in the code, but the error x and error y is a huge number , what should i do to avoid this peoblem? Thanks in advance!

ShingoMatsuura, 2017-03-07:
Did you set each length from anchor (0,0)?

I recommend that you check Pozyx tutorial 'Tutorial 2: Ready to localize' first.
[Tutorial 2: Ready to localize](https://www.pozyx.io/Documentation/Tutorials/ready_to_localize)

I hope it will help you.

OXINARF, 2017-03-08:
@walkeryu123 Have you read the [wiki article about it](http://ardupilot.org/copter/docs/common-pozyx.html)?

User questions should be posted in the forum (http://discuss.ardupilot.org). If this is a development question then please post in the forum or ask in Gitter (http://gitter.im/ArduPilot/ardupilot).

closed 2017-03-08 by OXINARF

rmackay9, 2017-03-08:
@walkeryu123, I'm keen to help you get this working though.  So please don't give up.  If you raise a discuss question feel free to post a link here so I see it.  If you and don't get an answer, please come find me on the gitter channel linked above.

RoboBil, 2017-11-28:
I've set up my rover with a POZYX tag and four anchors mounted in a not square configuration outside. My North-South  drive is tracked correctly with the mission planner but my East-West tracking is backwards. I can play with the BCN_ORIENT_YAW, but I'm still don't understand how the program interacts with the Pixhawk. It appears that the POZYX sends 2 groups of data; 1) distances from the tag to each anchor and 2) the position the tag per POZYXgetposition. 

Why send the individual anchor to tag distances to the Pixhawk when localization normally needs just getpositioning?