ArduPilot/ardupilot #5831: Plane: Maxbotix I2C rangefinder not working with PH2.1

cjneal, 2017-03-07:
#### Issue details
Have multiple airframes using Maxbotix I2C EZ4 sonars for auto landing.  Just converted to PH2.1 and am no longer getting any reading from the rangefinder.  Other I2C items (airspeed) work just fine, so I assume this is a software problem.  Have tried multiple planes and multiple sonar units.

#### Version
PH 2.1
AP 3.7.1
MP 1.3.44

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Multiple...pusher flying wing (X8), custom puller aircraft, and others

#### Hardware type
PH2.1

proficnc, 2017-03-07:
its possible that this driver is only checking the first I2C bus. 

that being said, I would not recommend a sonar on any airframe if you can avoid it.

magicrub, 2017-03-08:
@proficnc Looks like you're right. That driver seems to be hardcoded to a single i2c bus:

See example of a different driver correctly checking both buses - try 0 then try 1:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.cpp#L615

Maxbotix driver checking single bus - assume 1:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.cpp#L619
which calls
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_MaxsonarI2CXL.cpp#L38

cjneal, 2017-03-08:
@proficnc Why do you suggest avoiding sonar on planes?  I do understand why the lidar options are better in general, but for small planes where I want to avoid the weight and size I've had great luck as long as the landing params are tuned well.  Is there some reliability or performance reason?

proficnc, 2017-03-08:
If it's working for you, then great! But I find we have less complaints from lidar users...

magicrub, 2017-05-16:
looks like PR https://github.com/ArduPilot/ardupilot/pull/6260 may fix this. @jschall please confirm

bhollehday, 2017-05-17:
That would be awesome Im trying to get my maxbotic sonar's working again. I cant find a simple way of getting any obstacle avoidance read through AP.

cjneal, 2017-06-16:
@magicrub 
@jschall 

Is there a status update on PR #6260 ?  I'm looking for a version of Plane 3.7.1 with I2C working for either Maxbotix or (maybe preferably) Lightware SF11.  Is there someone who could build/compile a one-off variant of 3.7.1 with @jschall 's changes?

Tom-Shen, 2017-09-15:
Any conclusion? I have the same issue when I connected sonar to Pixhawk2

PR #7000 (2017-10-09): Lightware and Maxbotix I2C: search buses 0 and 1
After three PRs trying to fix Lightware detection in the Cube (#6558, #6710 and #6991) in the wrong way, I decided to do it myself with the best solution we have now - search buses 0 and 1 when a specific bus isn't specified. Hopefully we'll get to a point where we can get a list of I2C buses to search and we remove the macro.

@chobitsfan Can you please check this works for you?
@rmackay9 @mirkix If either of you can test on a BeagleBone board it would be great. I'm pretty sure I didn't break it, but I don't have hardware to test.

Also made a  [... cut]
closed 2017-10-09 by OXINARF