ArduPilot/ardupilot #5832: Copter: autotune to always use PosHold

magicrub, 2017-03-07:
When performing an autotune with GPS lock available always use PosHold. Else use AltHold.

With the newly added PosHold feature, there's no reason not to use it if we can.

geofrancis, 2017-03-07:
+1 i have always wanted this but i just assumed there was some technical reason it couldnt be done.

magicrub, 2017-03-08:
The PosHold feature was recently added via https://github.com/ArduPilot/ardupilot/pull/5468 but the current behavior is the autotune mode acts like whatever your previous mode was. I suggest we always act like PosHold if we can.

OXINARF, 2017-03-08:
I would argue otherwise. The current system allows anyone to choose between using auto positioning (it's more than just PosHold) or not, why would we reduce options?

magicrub, 2017-03-08:
>why would we reduce options

@OXINARF  Why would we do an Autotune without in PosHold if GPS is available? Can you give me a use case?

OXINARF, 2017-03-08:
Because you want to control it yourself instead of ArduPilot automatically controlling it? That's like asking why would you want to use AltHold when GPS is available...
Also since it tries to auto-yaw according to wind there is always the possibility that it's not taking the best decision and you would have to always be correcting it.

magicrub, 2017-03-08:
@OXINARF but you don't want to control it unless it gets off course.. which it won't if it's in PosHold! It allows for much much faster tunes. I don't understand your argument. Do you understand mine?

OXINARF, 2017-03-09:
@magicrub Even automatic control will only go until a certain limit, you can't say you never want to control it. My main issue isn't even with position control, it's that it also does automatic yaw rotation - and I doubt you can say it will do the right thing in every single environment.

I don't get what your issue is with having options. If you always want to use automatic control just initiate autotune from Loiter or PosHold - you should be in those methods already if automatic position control is important to you.

magicrub, 2017-03-09:
I'm not convinced. Doing an autotune from a mode other than PosHold is just a waste of battery.

rmackay9, 2017-03-09:
I think there are some use cases where people won't want the position hold.  It fits with our regular practice of pushing something out new functionality without changing the default behaviour.  I think we should see how it goes with both being available.  If users get confused and they always want to use poshold then we can make that the default then.  I suspect that what we really need is a better interface to the autotune feature.  I little screen in the ground station plus updates of it's progress.

rrr6399, 2017-03-09:
@magicrub, do you get about the same answers from autotune using PosHold and AltitudeHold? If so, PosHold would reduce the required continued manual recentering of the copter by the pilot.

magicrub, 2017-03-09:
I have not compared the results, but it certainly reduces the manual and recentering.

rmackay9, 2017-03-09:
@rrr6399, it should definitely produce the same (or slightly better) results.  The only reason it would be better is if there was a bit of wind because it rotates the copter so the twitches happen perpendicular to the wind.

geofrancis, 2017-03-10:
When was poshold auto tune introduced? I always thought it only worked with althold?

magicrub, 2017-03-10:
@geofrancis see PR https://github.com/ArduPilot/ardupilot/pull/5468

tridge, 2017-03-14:
I'd prefer to leave it as it is now, and make it depend on the flight mode.
It is actually theoretically better to autotune without poshold, just less convenient

magicrub, 2017-03-14:
The gods have spoken...

closed 2017-03-14 by magicrub