ArduPilot/ardupilot #5835: All Vehicles: The Pixhawk Slim FMU board AK8963 Compass fails to start

gmorph, 2017-03-08:
#### Issue details
The FMU board has an SPI connected 9250 IMU which has inside it an I2C connected AK8963.  The current start up code in all vehicle types tries to initialise the MAGs first followed by the IMU's.  That ordering doesn't work in this case as the the SPI bus needs to be brought up before we can talk to the I2C bus that the AK8963 is hanging off.

So when I startup a PHSlim with no other sensors/devices connected to it there is no operational MAG.

This might be desirable as on board MAG's functional poorly due to all the RF noise.

#### Version
Master

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Testing on a Pixhawk Slim which is the FMU board only out of a Pixhawk 2 - the vibration isolated board has been removed.  I suspect it doesn't start on a standard Pixhawk 2 either.

#### Logs

OXINARF, 2017-06-02:
@gmorph The AK8963, when in a MPU9250, is now forcing the IMU to be detected first. Can you confirm this is no longer an issue?

amilcarlucas, 2017-10-22:
@gmorph ping....

peterbarker, 2021-07-30:
Wasn't this fixed with the whole "AuxillaryBus" thing?

closed 2021-08-02 by tridge