ArduPilot/ardupilot #5840: AC 3.5 CAM Message enhancement

roque-canales, 2017-03-08:
for Copter 3.5 and RTK, it will be very useful to generate real position of camera by adding offset x y z parameter for CAM  beetween gps antenna centroid and camera sensor. 

Like this position of camera will be correct when multirotor lean angle.

mboland, 2017-03-08:
+1 on that
And can we have sub 1 second time stamps?

ptsneves, 2017-03-10:
@OXINARF , if I provide a patch for this will you accept the addition of this parameters?

ThorstenDP, 2017-03-10:
+1 and another +1 if we can get the "true" RelAlt above Home logged #4468

OXINARF, 2017-03-10:
@ptsneves I would probably ask for others' opinion (as I don't use the cam functionality myself) but this seems worth it to me.

rmackay9, 2017-03-11:
In reality I think the timing difference between when the camera message is sent and when the picture is actually taken has a much larger impact on the accuracy of the reported location than the normally small position offset on the vehicle.

Still, adding offset would be consistent with what we've done for other sensors (GPS, optical flow, precision landing camera) so I'm not against it.  One question though - should it be in the mount library instead of the camera library or are we talking about a hard mounted cameras only?

jmachuca77, 2017-03-11:
Its not as simple as just adding an offset to the position. The angle of the vehicle at the time the picture is taken has to be taken into consideration to obtain the correct position, this is called lever arm correction. If this is also contained in the PR then it would be a great addition. If the PR only adds an offset then It will not be accurate.

ptsneves, 2017-03-11:
Well the arm correction makes sense on the mount.
So adding offset and angles. The calculation of the arm is given by for instance tg(a)*offset_x. What is the angle convention here and the referential we should use?

ThorstenDP, 2017-03-11:
Using a stabilized gimbal the camera position has a continuoisly changing lever arm. But anyway it might help here as well since we have a better estimate of the average camera position.
I agree with Randy that the larger impact comes from the shutter delay. But if we have a precise hotshoe feedback (Trigger + CAM messages) and use RTK a camera offset would be really nice to have.

ptsneves, 2017-03-11:
How you do have a continuously changing lever arm if you take into account the pitch roll and yaw angles? Do you have telescopic mounts?

roque-canales, 2017-03-11:
Between RTK antenna gps centroid and camera, on most multicopter frame rtk antenna is above the CG of drone easily 10-15 centimeter, and camera is easily 10-15 centimeters below.

So with rtk + hot shoe, we can have centimetric position of antenna, but no of camera.
Adding x,y,z camera offset for trigonometric processing in live it with roll pitch yaw of multicopter frame, permit to maintain centimetric camera positions.

If not, with lean angle of 25° we add easily 10-15cm of camera position error, witch disable totally the interest of using rtk for georef images.

ThorstenDP, 2017-03-11:
@roque-canales yes right! But 10-15 cm is good enough for many applications and it is only possible with RTK/PPK as well. But then the camera offset is not crucial.
Not, sure but can we get the angles of the BGC? Maybe from Strom32?

roque-canales, 2017-03-11:
First resolve this for fixed gimbals. so only offset between rtk antenna and camera + roll pitch yaw are necessary to determinate real camera positions.

Second, if we use 2 axis stabilised gimbal, normaly for survey we put it in nadiral orientation. (-90).
So camera is always facing earth.

Also you can determine gimbal pitch orientation reading RC input value assigned to pitch and scaling it by reading MNT_RC_IN_TILT / MNT_ANGMIN_TIL / MNT_ANGMAX_TIL / RCx_MIN / RCx_MIN

ThorstenDP, 2017-03-11:
That's true! Depending on the camera there will be a difference between the COG of the gimbal and the lens but this is usually < 5cm.

OXINARF, 2017-03-12:
If we over complicate it gets harder to get merged. I think that accounting for the vehicle attitude is essential (as otherwise the offset won't have much value), but I think that a gimbal won't change the position that much to be essential to get done now.

ptsneves, 2017-03-12:
[See this commit branch from master for this issue](https://github.com/ptsneves/ardupilot/commit/8048329e951bd3840f63dd05b32700acd75326b1).
I have a patch where CAM messages now have the offsets of the mount included in the location calculation.

I have made a basic patch for ArduCopter and ArduPlane, but I am unhappy with the architecture of the new system.
In this new patch I added the offset parameters to the mount but it seems the mount is not very connected to the AP_Camera or camera messages and it results in an awkward system.
I have not tested this, it is just a proof of concept and compiles in ArduCopter and Arduplane.
[See this commit branch from master for this issue](https://github.com/ptsneves/ardupilot/commit/8048329e951bd3840f63dd05b32700acd75326b1).

I did not open a pull request yet because this code is not ready for merge. Let me know what you think.

ptsneves, 2017-03-12:
Another question is why is the AP_Camera not related to AP_Mount at all. A mount can have different things beside a picture camera but a picture camera for sure needs a mount, even if only a conceptual static mount, but one which has position (the optical center of the camera for correctly designed mounts).

This mess leads to the fact that my patch is incompatible with AP_Camera being enabled and AP_Mount disabled.

WickedShell, 2017-03-12:
@ptsneves You are missing altitude adjustments from pitch, and I don't see you actually apply R/P/Y corrections to the data anywhere there? location_offset() only translates the provided location by a number of meters, you need to externally sort out the effect of the attitude on this translation.

Your second _offset group is missing the 2 prepended to the parameter name, which means you have 2 parameters with the same name which isn't good, thats an easy fix though :)

Handling a lack of mount can be done by passing a null pointer and managing that where you calculate the offsets. Otherwise I suppose you could make an offset for a static mount.

OXINARF, 2017-03-13:
I don't think this should be in the AP_Mount library but instead in the AP_Camera one - it is defining the position of the camera, not the mount.

> Another question is why is the AP_Camera not related to AP_Mount at all

AP_Mount is probably misleading, it should be named AP_Gimbal, that's what it is for.

ptsneves, 2017-03-13:
@WickedShell Yes I have not finished the patch. The lady does not allow for much work on weekends. The attitude will be adjusted the same way as commented in libraries/AP_Mount/AP_Mount.cpp@539. The remaining angles will follow the same logic (I just need to verify the angle and referential convention).

Regarding the 2 offset groups with the same name, this was a good spot. I will fix it. On the topic, where on the code is the second mount used?

Your solution for lack of mount is also the most elegant given the situation. To const pointer we go!

@OXINARF I think this logic should not be in AP_Camera. As I said a Camera always is fixed on a mount even if a static/conceptual one. The current AP_Camera being decoupled from AP_Mount is something to be fixed, not create further wrong designs. Even Randy _seems_ to point to all mounted cameras, whether hard mounted or not. If the patch will not be accepted because of this I will not re-factor it and keep it in my fork, even though I would be disappointed.

I will try to update the patch set today.

OXINARF, 2017-03-14:
> As I said a Camera always is fixed on a mount even if a static/conceptual one.

Yes, but a static mount has no code to go with it, it's invisible to the autopilot.

> The current AP_Camera being decoupled from AP_Mount is something to be fixed, not create further wrong designs.

That's your opinion. It's not wrong in my opinion. You can have a camera without a gimbal and you can have a gimbal without a camera.

> If the patch will not be accepted because of this I will not re-factor it and keep it in my fork, even though I would be disappointed.

**I** wouldn't accept it, but others might. The proof to me that this is wrong is the simple fact that you are adding two sets of parameters (and methods that work with them), but you only use one of them since the AP_Camera library only supports one instance - and that shows this is AP_Camera related, not AP_Mount.

ptsneves, 2017-03-14:
@OXINARF By this order of ideas the frame type of a quadcopter would also be invisible to the Autopilot. After all it is just a static frame. Regardless of the approach, if you come up with an alternative implementation I do not thing choice would be a tragedy ;)

Back on topic: 
```
Matrix3f vehicle_dcm;
vehicle_dcm.from_euler(ahrs.roll, ahrs.pitch, ahrs.yaw);

Vector3f rotated_offsets = vehicle_dcm * offsets; 
location_offset(mount_location, rotated_offsets.x, rotated_offsets.y);
```
Is this the correct code to rotate the offset to a body centered coordinates?

ptsneves, 2017-03-14:
New commit with angles and refactor to use the AP_Camera instead of the AP_Mount. The way the ifdefs are set forces the usage of AP_Camera.
https://github.com/ptsneves/ardupilot/tree/camera_offset

OXINARF, 2017-03-15:
> By this order of ideas the frame type of a quadcopter would also be invisible to the Autopilot. After all it is just a static frame

In a sense it is. We only need to know how many motors you have and where they are because of physics - but then all the code is the same for a quad, a hex or an octa. But where your comparison fails is that the frame isn't static, it's moving in the air, if it was static you wouldn't need an autopilot :smiley: A static mount on the other hand isn't moving (the frame is) so it doesn't matter to the autopilot.

Anyway, new code looks better to me - a couple of minor issues here and there, but very easily fixable. Regarding the offset math I'm not the person to comment on it. @WickedShell can you help here?

ptsneves, 2017-03-15:
@OXINARF You have a point. Can you point out the minor issues?
@WickedShell Can you give a look at the math code?

Also I have updated the branch to use the offsets in the mavlink message. I have not added this offset calculation to the update_location because the navigation is done in terms of cg position and with a trigger based on offsets, this could create unreachable triggers.

WickedShell, 2017-03-15:
@ptsneves I don't really have time to work out the correct math at moment, but the proposed math isn't correct, as it doesn't apply any altitude compensation (IE from pitching moments).

ptsneves, 2017-03-15:
I think you are referring the my first patch. My recent one uses the dcm matrix from the yaw,pitch and roll to rotate the point. Can you confirm this does not compensate for the pitching moments?

WickedShell, 2017-03-15:
I'm not prepared to comment on the DCM part, but you never actually touch the altitude when you do the corrections.

ptsneves, 2017-03-15:
Dhough you are right. I was thinking that the rotation matrix was enough but this only calculates the correct components. I adde the offsets to the horizontal coordinates but forgot adding the z component.
I already pushed the fix.
Is there anyway I can validate the math?

OXINARF, 2017-03-16:
@ptsneves I've commented on your commits with the minor issues I've seen. If you aren't certain of the math, the best is to open a PR and I'll try to get someone to review.

roque-canales, 2017-05-04:
Does someone pushed this into master for it be released with the next beta for we test it?

ptsneves, 2017-05-04:
@roque-canales It was my fault. I forgot to open the pull request as OXINARF requested. I will try to do that today.

roque-canales, 2017-05-13:
@Ptsneves don't worry is just that 3.5 rc6 will be release in the next few days and maybe it can be added in ?

roque-canales, 2017-06-10:
Hello Randy,

Could you please merge this PR at master for it be added to 3.5 release?
https://github.com/ArduPilot/ardupilot/pull/6266

roque-canales, 2018-01-06:
Hello, there is update on this feature? It will be avalable for test in next beta release?

norimboo, 2019-03-05:
Hi what is happening with this update ? When it will be released ?