ArduPilot/ardupilot #5846: EK3 GPS FLOW RNG switching issue

roque-canales, 2017-03-09:
Hello Paul,

After many test, I can tell you that:
When we start from no gps zone, ek3 use flow/rng well,
During flight when we go to gps zone, ek3 switch to use GPS and fusion with flow/rng.
But when we return to no gps zone, even if we only have 4 sat an poor precision and lot of reflections, EK3 no return to only use flow/rng as when we start.

This issue finish all time by EKF ERROR.

All seems to work as once gps lock is acquired, EKF use gps input even is really poor.

Could you please make that in this situation, EK3 only use rng/flow and reject gps (led blue)???

OXINARF, 2017-03-09:
@roque-canales If you are implying there is a bug in EKF that doesn't allow going from using GPS to using optical flow, then please provide a log.

rmackay9, 2017-03-09:
Great that you're testing EKF3!

roque-canales, 2017-03-10:
[93.zip](https://github.com/ArduPilot/ardupilot/files/834091/93.zip)
I'm implying that:
Going from only optical flow (no gps zone) to optical flow fusion with gps on gps zone it's ok.
Going from optical flow fusion with gps (on gps zone) to no gps zone with flow it's not ok.

Effectively on startup on no gps zone, GPS module cannot track sat and wait to receive good quality signal.
But going for good reception GPS zone to no gps zone (indoors or very close to walls outdoors) no work well, because gps module continue to track poor signal with lot of reflections and sends out bag gps pos en vel estimations....

Due to this GPS behaviour, returning from gps fusion with optflow to indoor (no gps zone) goes bad.

We need to simply reject gps reading from EKF reading in this case for example doing as for ahrs, rejecting gps if we not reach 6 sats.

Today this behaviour make EKF error, due to very bad vel/pos innovations from GPS.

I tryed to tune ekf vel/pos gates by reducing it gradualy, and effectively EKF no use bad readings regarding gates but reducing gates, ekf goes into failsafe mode as soon as the gps sends out bad readings, instead of switching to a no longer listens to the gps and lissen only flow.

HockyYoung, 2017-04-08:
Did you use PX4Flow?,now I use firmware of 3.5.0-rc3 ,I change AHRS_EKF_TYPE=3  EK2_ENABLE=0 EK3_ENABLE =1, EK3_GPS_TYPE=3,FLOW_ENBALE =1,When I use loiter mode to disarm,but the copter could not work.Is the copter 3.5.0 not support PX4Flow？ @rmackay9 @roque-canales

priseborough, 2017-04-16:
The EKF de-weights the GPS according to it's reported accuracy and the normalised innovation check levels. However there are a couple of issues with your log that stand out

1) There is a bug with the GPS driver that is causing the reported horizontal position accuracy to wrap back to zero - this is potentially very bad for the EKF. @tridge FYI

[image]

2) The range finder observations are very noisy for the second flight and range finder data gets rejected by the EKF for several seconds at a time. @roque-canales what surface were you flying over and what range finder were you using?

[image]

It would be to advisable to implement a different minimum GPS quality threshold when there are other forms of navigation available as we can afford to give up on GPS earlier. This should be based on the receivers reported velocity accuracy in combination with other checks (the number of satellites on its own is not  adequate). This  is something I will have a look at.

WickedShell, 2017-04-16:
@priseborough all the bad GPA messages are associated with a period where the GPS.Status is NO_FIX so the accuracy should probably be disregarded. I suspect we have a data storage problem there, I will look into it though, I think I know why it rolls over.

WickedShell, 2017-04-16:
@priseborough accuracies come in as 32 bit values, and we only log them as 16 bit values, you are seeing the warp associated with the truncation. The actual value is still going up, this isn't a flight critical bug, rather its just a logging one.

roque-canales, 2017-04-18:
@Paul thank you for your report regarding theses logs.
Yes I like your proposition to weight differently gps when other sensor is available and also weigh it regarding gpa Hacc Sacc rather than gps sat available.

Regarding rangefinder is an sf11c above water and wet road.

priseborough, 2017-04-18:
My limited experience with optical flow above water is that it is not reliable - was the second flight with the noisy range finder data above water?

roque-canales, 2017-04-18:
On this log, second flight was only above wet road near wall.  So no flight above water.

IamPete1, 2021-09-11:
Any updates? Is this still a issue?

IamPete1, 2021-09-22:
No update in 4 years, I'm going to close. Please reopen if this issue re-appears on 4.1.

closed 2021-09-22 by IamPete1