ArduPilot/ardupilot #5856: Feature request: Enhance existingt lost model alarm functionality by adding support for loud lost model / signal lost alarms

Saxin, 2017-03-11:
#### Issue details
Pixhawk buzzer volume is too low to be heard when looking for lost model.
Products providing load beeping capabilities exist and can be classified as follows:
1. Simple two wire active buzzers -
[http://www.banggood.com/5V-Active-Buzzer-Alarm-Beeper-With-Cable-for-FPV-Racer-Quadcopter-Drone-DIY-p-1032123.html?rmmds=myorder](url)
2. Single function lost model buzzers - 
http://www.banggood.com/Matek-Lost-Model-Beeper-Flight-Controller-5V-Loud-Buzzer-Built-in-MCU-for-FPV-Multicopters-p-1106877.html?rmmds=search
3. Multi function usually battery voltage, lost model and activity buzzers - http://www.banggood.com/Matek-Signal-Loss-Alarm-Loss-Aircraft-Finder-with-Lipo-Display-p-943085.html?rmmds=search
(Links to BG only due to convience, I'm not afilated with them in any way :))

Most of users today use PPM / Serial connections between FC and RX, plugging into RX isn't an option.
Due to https://github.com/ArduPilot/ardupilot/issues/1235 & https://github.com/ArduPilot/ardupilot/issues/1239 which according to @rmackay9 can't be resolved, some of those buzzers work not as intended when connected to Pixhawk servo out + pass through configured. 

A big benefit here is allowing users with radio systems without free channels (or maybe with less channels) still use a loud lost model alarm without the need of dedicated TX channel. This should be of course added to the wiki to help new users with setup.

It would be nice to add support for some common types of buzzers connected to one of the servo channels being an expansion to lost model capability, for example setting SERVO9_FUNCTION to 80 (new value). 
Additional parameter may be required to configure required behavior on that out, for example unless LOST COPTER FUNCTION active, 
a. toggle low - high [in PWM sense] every 1 sec,
b.toggle low, 
c. toggle high,
d. or something else.

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk

geofrancis, 2017-03-13:
+1 this would be useful for me as the pixhawk lite doesnt have a buzzer

peterbarker, 2020-06-03:
My understanding of this issue is that we want support for the really loud buzzers you can attach to the servo pins.

I believe we already have that; the existing notify buzzer would appear to allow you to set a servo pin as an output.

Note that we've been able to do this since forever, so I might be missing something.

This buzzer should already be being used for the lost-vehicle alarm.  Triggering the lost vehicle alarm for a crashed vehicle is a separate issue.