ArduPilot/ardupilot #5859: RC Inputs & Outputs freeze randomly (crashlog)

ClementChristomanos, 2017-03-13:
#### Issue details
This issue has been reproduced on two different airframes that share the same configuration.
While in flight, all RC inputs & outputs freeze - no pwm is given to the ESC, thus the drone crashes.

#### Version
ArduCopter 3.4.5

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Both lost airframes were hexa. 

#### Hardware type
Pixhawk with companion computer over usb. 

#### Logs

[161.BIN.zip](https://github.com/ArduPilot/ardupilot/files/839301/161.BIN.zip)
[image]

On that last image you can see pretty clearly all RC Ouputs go to blank - VServo as well. 
Also, it looks like Performance monitoring stopped logging a few ms before the crash event.

Thanks for your help !

kd0aij, 2017-03-13:
I thought this kind of hang was likely with USB connected (the PX4 prearm checks say it's unsafe), but can't find anything relevant in the forums/docs except for this unanswered question: https://groups.google.com/forum/#!topic/px4users/ke2xiGMMSOY

ClementChristomanos, 2017-03-13:
Thank you for your input. Indeed USB is connected, and @rmckay9 already pointed me on the fact that USB might be unsafe in flight.
However I would like some more information regarding that statement since I've flown on a daily basis for the past two years and a half with USB connected without any similar issue. Unless there's been recent modifications to the USB drivers I doubt that this is the root cause of the issue.

ClementChristomanos, 2017-03-17:
Any more inputs on this ?

ClementChristomanos, 2017-04-06:
This bug is a potential danger to arducopter users using 3.4.5 and there's no clue whether this bug affects further versions or not. @rmackay9 @tridge @OXINARF any inputs on these logs ?

Clément.

rmackay9, 2017-04-21:
Hi Clement.
Sorry for not seeing this report.
I've had a look at the logs and, as you say, the RCIN and RCOUT suddenly flat line and then the vehicle loses control.  I've had a look at the logs and I don't yet see the cause.  Tridge is going to have a look as well to see if he has any ideas.  Here are some things that could be a good starting place for our investigation:

- the USB was plugged into the flight controller (I assume from the comments above)
- LidarLitev2 is being used (a badly behaved sensor but I've never personally heard of it affecting RC)
- motor-compass compensation using the throttle method is being used (this is *not* a good method because the interference is generally not linear with throttle).  Looks like you have a current monitor so that would be a better method although if you have an external compass this compensation is really not necessary.  It's hard to imagine how this could be related though.
- servo gimbal being used.  The servos are powered from the rear rail of the pixhawk?

tridge, 2017-04-21:
> Interestingly the POWR.VServo flat lines at the same time as the RCIN, RCOU. This seems like an important clue.

yes, looks like the PX4IO microcontroller stopped communicating with the FMU.

Clement, what type of RC input system do you have? What brand and what protocol is used to connect it to the pixhawk?

tridge, 2017-04-21:
also, do you have the log for the 2nd crash?

OXINARF, 2017-05-02:
@ClementChristomanos  We talked about this in dev call today. This is very hard for us to debug because this happens only to two users (assuming it is the same issue as #6127) and not to any of us - the fact that there is no logging or anything post-mortem from the I/O CPU also doesn't help as we only know the link was lost. What we would like to ask you is to try and reliably reproduce it **on the bench**. If you can do that and give us a list of steps to make it happen, then we'll ask someone on our team to reproduce it in our hardware - hopefully we don't need to get to the point where we need your hardware to reproduce it.

I'd like to point out that this is on our list and a big worry to us, but our hands are kind of hands tied without being able to reproduce.

khancyr, 2017-05-02:
Hello,
Which pixhawk hardware were you using ? Dropix version ?

ClementChristomanos, 2017-05-02:
Hi, Thank you for your help ! Here are some answers.
- Indeed, LidarLite v2 was used during this flight. To me this is unlikely to make the whole RC channels freeze. We experienced some bugs before, the usual "13m offset" bug, but nothing else.
- About the current sensor / throttle motor compass compensation, we found the current sensor to be unreliable and preferred to switch to throttle compensation. We are currently investigating on getting more reliable current sensing so we can switch to a more appropriate compensation method.
- A gimbal is used indeed but this is not relevant as the first crash happened on a different drone without any gimbal. No other elements are wired to the pixhawk servo outputs.
- @tridge The drone itself is controlled through a proprietary companion computer and associated software, and also controlled through a standard ppm Frsky transmitter. At the time of the crash, the companion computer was in control in guided mode.
- Indeed, the logs make it pretty clear the PX4IO microcontroller and FMU stopped talking to eacher other.
- Unfortunately the SD card was completely destroyed from the second crash. 

- @OXINARF We tried multiple times reproducing this on a bench but could not reproduce it so far. I'll let you know if we see the same behaviour again. 
- @khancyr Indeed we use a Dropix version, which seems to be also the case with #6127.

I'll let you know whether we get more data regarding this issue. But more than finding the underlying problem I think this could lead to having a better redundancy strategy with the IO processor - perhaps the IO processor could send out PWM to the ESCs in an emergency mode instead of cutting all signals to ESCs. 

Thanks for your help !

Josephthomas17, 2018-09-07:
Hi team, 

Had my Y6 crash some weeks back with the same flat lines (attached .BIN file).

APM:Coper V3.3
No companion computer like Clement's case

[Crash .BIN file.zip](https://github.com/ArduPilot/ardupilot/files/2360685/Crash.BIN.file.zip)

[image]

Would be great to hear about the insights if the issue was fixed. Thanks for your time. 

Regards,
Joseph

khancyr, 2018-09-07:
@Josephthomas17 
Hello, 
Please open a post on http://discuss.ardupilot.org/ . And you are using copter 3.3 so if there was a bug, it was surely found long time ago ..

peterbarker, 2021-07-30:
No response in 3-odd years, closing.  Been a lot of code under the bridge since this was raised.

closed 2021-07-30 by peterbarker