ArduPilot/ardupilot #5860: Detect a unhealthy airspeed sensor bus

WickedShell, 2017-03-13:
#### Issue details
Sudden large step shifts of temperature data can be used to detect a corrupt and bad transaction with the airspeed sensor. We can then flag the sensor as unhealthy as at least a first step.
[image]

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Logs

[19.zip](https://github.com/ArduPilot/ardupilot/files/839415/19.zip)

OXINARF, 2017-03-14:
Interesting. One thing to be careful is if this happens with all sensors - if it does it's possible to have the frontend deal with it, otherwise it needs to be the backend of the ones where this happens.

WickedShell, 2017-03-14:
My assumption was to manage this at the backend level, for at least the MS4525DO driver. I don't think generalizing that all sensors would have the same behavior is a particularly decent idea... But I could be convinced otherwise, at the moment it just felt more like a per driver deal to me.

magicrub, 2017-03-14:
I agree the failure is probably different per driver so having it in the backend is best. Would be nice to have a pure virtual function to ask if it's acting funny.. or maybe it just handles its own healthy flag?

magicrub, 2017-03-14:
Also, this is definitely a duplicate.

WickedShell, 2017-03-14:
Yeah I know there was an issue about falling back, I only opened this as a specific type of health failure we can detect. Feel free to squash them if you want :)

magicrub, 2017-03-14:
At least link to the other related open/closed issues/PRs. I've authored a few of each on this topic. If I wasn't mobile I'd do it myself.

WickedShell, 2017-06-03:
We preform double reads and sanity check that the readings are reasonably close to each other, otherwise we discard the readings as corrupted. This accomplishes the behavior this issue was asking for. This was done as #5950

closed 2017-06-03 by WickedShell