ArduPilot/ardupilot #5877: Plane: QuadPlane should handle yaw reset like Copter

rmackay9, 2017-03-15:
The EKF can reset it's yaw when it climbs about 2m after takeoff (at least it does in Copter).  Assuming this reset happens when the EKF is being used in QuadPlane then we should reset the yaw target like we do in copter.

The relevant copter function is "check_ekf_reset" in ekf_check.cpp.

DavidIngraham, 2018-01-09:
+1
We had a vehicle with bad compass interference where this caused big problems. The yaw would reset on every takeoff, leading to a very scary uncontrolled yawing motion.

mwingerson, 2019-01-17:
has this been addressed?

prateekjain0608, 2020-09-10:
Has this been addressed?
I am having a similar issue in Tailsitter. @rmackay9 
Immediately after take-off, MP HUD displays "Error Compass Variance" and there's "EKF Yaw Reset" messgage.

moreba1, 2025-01-30:
Hi
does this feature implemented ?