ArduPilot/ardupilot #5881: error about sim_vehicle.py

leesf1996, 2017-03-15:
Hello!
I'm working on [Setting up SITL on Linux](http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#setting-up-sitl-on-linux).When I started SITL simulator:

sim_vehicle.py -w

Things went wrong:

SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/sophie/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/sophie/ardupilot/Tools/autotest/sim_vehicle.py", line 980, in <module>
    do_build(vehicle_dir, cmd_opts, frame_infos)
  File "/home/sophie/ardupilot/Tools/autotest/sim_vehicle.py", line 566, in do_build
    return do_build_waf(opts, frame_options)
  File "/home/sophie/ardupilot/Tools/autotest/sim_vehicle.py", line 532, in do_build_waf
    run_cmd_blocking("Configure waf", cmd_configure, check=True)
  File "/home/sophie/ardupilot/Tools/autotest/sim_vehicle.py", line 645, in run_cmd_blocking
    p = subprocess.Popen(cmd, **kw)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
SIM_VEHICLE: Killing tasks

Can anybody help me?Thanks!

khancyr, 2017-03-15:
Hi, user question should be post on http://discuss.ardupilot.org/ . Those issues are for developpement issues.
Otherwise, you must have miss a step on installation (probably git submodule update --init --recursive)

closed 2017-03-16 by leesf1996

peterbarker, 2017-03-17:
On Wed, 15 Mar 2017, leesf1996 wrote:

> SIM_VEHICLE: "/home/sophie/ardupilot/modules/waf/waf-light" "configure"
> "--board" "sitl"
> Traceback (most recent call last):
.
.
> OSError: [Errno 2] No such file or directory

> Can anybody help me?Thanks!

You need to run `git submodule update --init` - waf-light is not available 
there.

Did the instructions you're following omit that step?

borhanreo, 2019-05-30:
I faced the same problem. I install step by step but could not work. showing same error. When i restart my ubuntu pc then it could work.

vaibhav8851, 2021-10-03:
I am also getting the same error while starting sitl  on windows after installation I get 
this message 
$ ../Tools/autotest/sim_vehicle.py --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/hp/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/hp/ardupilot/modules/waf/waflib/Node.py", line 579, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/hp/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/hp/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/hp/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/hp/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/hp/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/hp/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/hp/ardupilot/wscript", line 49, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/hp/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/hp/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/hp/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/hp/ardupilot/modules/waf/waflib/Node.py", line 672, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/hp/ardupilot/modules/waf/waflib/Node.py", line 672, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'

can anyone help

rajpal007, 2022-01-24:
$ ../Tools/autotest/sim_vehicle.py --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/aditi/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/aditi/ardupilot/modules/waf/waflib/Node.py", line 579, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/aditi/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/aditi/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/aditi/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/aditi/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/aditi/ardupilot/wscript", line 49, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/aditi/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/aditi/ardupilot/modules/waf/waflib/Node.py", line 672, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/aditi/ardupilot/modules/waf/waflib/Node.py", line 672, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'

Please help me i read 100 fourms can none helped me . i even uninstalled python coz some said its python version issue . please help

magicrub, 2022-01-24:
For Windows I highly recommend using WSL.

peterbarker, 2022-01-25:
> SIM_VEHICLE: Killing tasks SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'

That's a Python2 vs Python3 issue.

*Are* you running under Cygwin, which @magicrub is guessing you are?

We can debug this if you can run branches as I push them up.....

peterbarker, 2022-01-25:
@rajpal007 please raise another issue which mentions which Windows and/or Cygwin....

mahimpat, 2022-07-06:
$ ../Tools/autotest/sim_vehicle.py
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/mahimpatil/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Node.py", line 579, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/mahimpatil/ardupilot/wscript", line 49, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Node.py", line 672, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/mahimpatil/ardupilot/modules/waf/waflib/Node.py", line 672, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'

tried everything still not resolved 
please help

magicrub, 2022-07-09:
There have been two suggestions: WSL and/or python 3. Have you tried either of those yet?

harouneerachide, 2024-06-28:
i am using ubuntu 20.04 
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/rachideu/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 590, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/rachideu/ardupilot/wscript", line 52, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/rachideu/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 683, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 683, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks