ArduPilot/ardupilot #5884: Ardupilot does not respond to TIMESYNC mavlink messages

fnoop, 2017-03-16:
#### Issue details
Ardupilot should respond to TIMESYNC messages by signing the message with the current timestamp and sending it back.  This is vital for synchronising timestamps between the flight controller and other components.

#### Version
master (3.5)

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
n/a

#### Hardware type
SITL

#### Logs
n/a

rmackay9, 2017-03-16:
Sounds good.. and doesn't sound too hard really.

ptsneves, 2017-03-17:
What does it mean to sign the message? To answer?

closed 2017-07-19 by fnoop

khancyr, 2017-07-19:
@fnoop can you let a message on why you are closing it ?

fnoop, 2017-07-19:
Oops sorry, 'cos it was implemented :)
https://github.com/ArduPilot/ardupilot/pull/5944