ArduPilot/ardupilot #5894: Coordinated Turn CH7/CH8 option - Alternative approach to Drift Mode

Pedals2Paddles, 2017-03-17:
#### Issue details
I was thinking this might kill a bunch of metaphorical birds with one stone on Copter.  Drift mode is awesome, but is limited to exactly what it is now. Basically stabilize mode with coordinated turns (yaw mixing).  The coordinated turns are an amazing effect for FPV and orientation. But you still need to control altitude and holding of position like you do in Stabilize. Not that this is bad. It's fun. But coordinated turns with altitude hold would be spectacular. Coordinated turns with position hold would be epic.  Coordinated turns in loiter would be very useful too. Maybe even some applicability in Sport mode.

Rather than creating a bunch of new flight modes, maybe it can be a CH7/CH8 option?  Call it something like "Drift Turns".  Then the operator can flip the functionality on for any applicable flight mode they want.  It should probably be inhibited in Acro, brake, auto, autotune, guided, RTL, land, takeoff, and simple modes.

This would allow every mission type, from fun to racing to professional production to make use of the coordinated turn functionality in the most effective manner.  And it minimizes flight mode clutter.

#### Version
Copter 3.5 onward

#### Platform
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All copter

#### Hardware type
Any

#### Logs
N/A

lvale, 2017-03-19:
As a flight mode modifier, how would it behave if another flight mode modifier is active. Example: Either Simple mode or Super Simple mode can be enabled. If we add "drift" as a modifier what happens to Simple/Super Simple if enabled too ?

Pedals2Paddles, 2017-03-19:
@lvale I am not sure. To be honest, I haven't used Simple or Super Simple in probably 3 years so I can only speculate. I think it would work just the same. It's just mixing yaw and roll. So it would be no different than if you pushed some right roll stick and some right yaw stick.  I don't believe it would be incompatible or need to be in be inhibited.  I think.

closed 2017-12-18 by Pedals2Paddles