ArduPilot/ardupilot #5896: AP_Baro temperature limiting

EShamaev, 2017-03-18:
#### Issue details
I am concerned about temperature limiting at 25 degree Celsius.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro.cpp#L337

Wrong temperature can lead to offset as described in comment, but also limiting to 25 when it is actually much higher can also give same problems.
The idea to use separate set_external_temperature for this is understood, however there is no single place where it gets called. So it will never get filled and the limit of 25 will work all the time.

#### Platform
[*] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

WickedShell, 2017-03-18:
> libraries/AP_Baro/AP_Baro.cpp#L337

set_external_temperature() is called to pass the airspeed temperature in. I
do however agree that we need more options. Tridge and I were discussing
that calling that with the temp of an external mag if available would be a
good addition.

The clamping is due to a very high reading on the baro that's usually
incredibly off from the real temperature.

EShamaev, 2017-03-18:
Where does set_external_temperature gets called? I could not find a single one.

WickedShell, 2017-03-18:
@EShamaev https://github.com/ArduPilot/ardupilot/blob/2b43c6aeedbd8bf4c601b8c765975bef585c783c/ArduPlane/sensors.cpp#L94

WickedShell, 2017-03-18:
That's a good point, we will need to update the interfaces to track where our temp source came from... (Airspeed -> ext baro -> compass is probably the best priority for those)

EShamaev, 2017-03-18:
But for sure not in ArduPlane.
We just discussed it with Tridge on UAVCAN channel and it should be done with parameters in libs. I fully agree to this.
Actually support for a normal OAT sensor would be nice too.

EShamaev, 2017-03-19:
I close this for now, as https://github.com/ArduPilot/ardupilot/issues/5906 is opened

closed 2017-03-19 by EShamaev