ArduPilot/ardupilot #5902: Send Optical Flow to ArduCopter through Mavlink

Ahrovan, 2017-03-19:
I check PX4 github, see below code, but arducopter dont have any code in Optical flow library for recieve Mavlink OPTICAL_FLOW_RAD & HIL_OPTICAL_FLOW  :

MavlinkReceiver::handle_message_optical_flow_rad(mavlink_message_t *msg)
{
  /* optical flow */
  mavlink_optical_flow_rad_t flow;
  mavlink_msg_optical_flow_rad_decode(msg, &flow);

  enum Rotation flow_rot;
  param_get(param_find("SENS_FLOW_ROT"), &flow_rot);

  struct optical_flow_s f = {};

  f.timestamp = flow.time_usec;
  f.integration_timespan = flow.integration_time_us;
  f.pixel_flow_x_integral = flow.integrated_x;
  f.pixel_flow_y_integral = flow.integrated_y;
  f.gyro_x_rate_integral = flow.integrated_xgyro;
  f.gyro_y_rate_integral = flow.integrated_ygyro;
  f.gyro_z_rate_integral = flow.integrated_zgyro;
  f.time_since_last_sonar_update = flow.time_delta_distance_us;
  f.ground_distance_m = flow.distance;
  f.quality = flow.quality;
  f.sensor_id = flow.sensor_id;
  f.gyro_temperature = flow.temperature;

  /* rotate measurements according to parameter */
  float zeroval = 0.0f;
  rotate_3f(flow_rot, f.pixel_flow_x_integral, f.pixel_flow_y_integral, zeroval);
  rotate_3f(flow_rot, f.gyro_x_rate_integral, f.gyro_y_rate_integral, f.gyro_z_rate_integral);

  if (_flow_pub == nullptr) {
    _flow_pub = orb_advertise(ORB_ID(optical_flow), &f);

  } else {
    orb_publish(ORB_ID(optical_flow), _flow_pub, &f);
  }

  /* Use distance value for distance sensor topic */
  struct distance_sensor_s d = {};

  if (flow.distance > 0.0f) { // negative values signal invalid data
    d.timestamp = flow.integration_time_us * 1000; /* ms to us */
    d.min_distance = 0.3f;
    d.max_distance = 5.0f;
    d.current_distance = flow.distance; /* both are in m */
    d.type = 1;
    d.id = MAV_DISTANCE_SENSOR_ULTRASOUND;
    d.orientation = 8;
    d.covariance = 0.0;

    if (_flow_distance_sensor_pub == nullptr) {
      _flow_distance_sensor_pub = orb_advertise_multi(ORB_ID(distance_sensor), &d,
                &_orb_class_instance, ORB_PRIO_HIGH);

    } else {
      orb_publish(ORB_ID(distance_sensor), _flow_distance_sensor_pub, &d);
    }
  }
}

Ahrovan, 2017-03-21:
add enum OpticalFlow_Type {} or edit OpticalFlow::init function for use Mavlink massage for front-end of \AP_OpticalFlow lib.
like \AP_RangeFinder library we can add AP_OpticalFlow _MAVLink libraries
some one give some addation info for do this Enhancement. thx

HockyYoung, 2017-04-08:
Ardupilot Firmware can alse support PX4Flow now? I use ArduCopter 3.6 to use PX4flow  ,and I found it can't loiter only with PX4Flow,can everyone know what's the problem.

Ahrovan, 2017-04-08:
@HockyYoung  see http://ardupilot.org/copter/docs/common-px4flow-overview.html

HockyYoung, 2017-04-11:
Thanks a lot. I've already use my px4flow to achieve loiter now.but the alt is not so well,I think I shoud debug my althold first.

bys1123, 2019-05-04:
http://ardupilot.org/copter/docs/common-openmv-optflow.html
I think now it's supported

rmackay9, 2019-05-06:
Yes, it's in matser now so I'll close this, thanks @bys1123

closed 2019-05-06 by rmackay9

spdrone, 2021-04-23:
> Ardupilot Firmware can alse support PX4Flow now? I use ArduCopter 3.6 to use PX4flow ,and I found it can't loiter only with PX4Flow,can everyone know what's the problem.

need a downward facing distance sensor