ArduPilot/ardupilot #5906: Air Data unified interface library

EShamaev, 2017-03-19:
Currently there is a mixture of different parameters from different sensors scattered among libraries.
Temperature comes to Baro. Baro does some calculation, Airspeed does some calculation and so on.

Proposed is a major rework of Air Data handling.
1. Separation of different sensors that really can be different.
2. Aggregation of all Air Data in one top class.

```
AP_AirData
+- AP_Static_Pressure
|    +- AP_BarometerSensors
+- AP_Diff_Pressure
|    +- AP_DiffPressureSensors
+- AP_OAT
|    +- AP_TemperatureSensors
+- AP_AOA
     +- AP_HardwareSensors or SoftwareCalculations
```
AirData can then also can change the plane's rudder sensitivity, aerelon sensitivity, etc to extend the safe flight envelope.
The number of each sensors might be more than 1 and there should be averaging/selection/blending setup as well as type selection.

A major rework is needed for this but it would definitely help in the long run.

tridge, 2017-03-19:
note that we already do control surface speed scaling based on airspeed. We use synthetic airspeed if we don't have airspeed sensor

tridge, 2017-03-19:
plus temperature could come from other sensors, eg. most external magnetometers have a temp sensor

EShamaev, 2017-03-19:
Speed: Unified class can do that.
Temperature: So there should be a way to select source closer to OAT.

tridge, 2017-03-19:
yes. I have no objection to the general idea, just commenting on this "AirData can then also can change the plane's rudder sensitivity, aerelon sensitivity, etc to extend the safe flight envelope." which seems to imply we don't do that now, which we do. We'd fly pretty badly if we didn't do that!

EShamaev, 2017-03-19:
Yes, we do, I know!
Could do better if we integrate full flight envelope and not only speed.

closed 2017-05-05 by EShamaev

DavidIngraham, 2017-11-30:
@EShamaev It seems like there was a desire for a user-selectable OAT source that is no longer captured now that this is closed. Do we want a separate issue for that?