ArduPilot/ardupilot #5913: Copter v3.4.5 & v3.4.6 Motor #4 anomaly:  Motor works with OctaX but not QuadX firmware

JStevens812, 2017-03-20:
#### Issue details
_Motor no. 4 will not arm or activate during any attempts at arming and testing motor setup in any of the copter firmware tested per this discussion:

<http://discuss.ardupilot.org/t/v3-4-5-motor-4-anomaly-fails-to-spin-up-when-armed/15709?u=eng6266>

When OctaX firmware is loaded, all four motors arm and spin up as desired.  Have tested all connections, wiring, hardware, and configurations to rule out possible known issues, but none have been the culprit._

#### Version
_3.2.1, 3.4.5, & 3.4.6_

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_Quad X_

#### Hardware type
_Pixhack v2.8.1_

#### Logs
_4 logs may be found in above link.  If additional logs are required, please let me know so I may upload them as soon as possible_

rmackay9, 2017-03-21:
I suspect this is actually a hardware issue, it's not a confirmed software bug so technically it doesn't belong on the github's issue list yet. Still, we can leave it here until we get to the bottom of the cause.

mboland, 2017-03-21:
We did try to investigate all the options in the support forum.
I don't know enough about the board design to be able to reason why an output would not work with one version of firmware and work with another version and still be a hardware issue.
It was I who suggested escalating it to the issues list.

rmackay9, 2017-03-21:
Yes, it's fine.  Thanks for your help digging into this issue.
I've posted some further tests for Joshua to try - those may help identify the issue.

amilcarlucas, 2017-10-22:
Try recalibrating the ESCs. We had a similar problem, one octa motor did not spin, re-calibrating the ESCs fixed it.

peterbarker, 2017-12-19:
@mboland did this get resolved?

JStevens812, 2017-12-19:
@peterbarker We never did resolve this issue.  I tried all of the tests @rmackay9 posted as well as the others previously listed with no luck.

@amilcarlucas I never had a problem with the Octa firmware spinning all 4 motors.  My issue was when I reloaded the Quad firmware (which is what I was actually flying) only 3 of the 4 would spin up.  The power package was working well before I swapped the Naza v1 flight controller out for the Pixhack.  I have since swapped in a Pixhawk 2.1 and everything is working perfectly.

I sold the Pixhack to someone willing to dig deeper into the hardware side of things, but have not heard back from them on whether they got the unit working.

peterbarker, 2017-12-19:
Notionally it could be a HW fault - but it would have to be a *very* clever HW fault!  Something like a timer only being initialised when more outputs are needed... or something.  Still, with the number of units produced, maybe you just lucked out.

I don't think anybody will ever, ever be able to reproduce this.  OK to close this issue?

JStevens812, 2017-12-20:
That's what I was thinking and I fully explained the situation and provided links to this forum as well as the RC Groups forum we originally started troubleshooting on to the person that purchased it from me, so hopefully they have the time and knowledge to figure it out or use it for something else entirely where there  isn't the conflict... who knows. 

Yes, okay to close this issue.  Thank you all for your help.

peterbarker, 2017-12-21:
@JStevens812 Thanks!  Hopefully the person doesn't have the issue or can get back to us.  It's an intriguing one!

closed 2017-12-21 by peterbarker