ArduPilot/ardupilot #5916: can't make px4-v2 in the window toolchain v16

QiMo2016, 2017-03-21:
#### Issue details
I‘ve got below messages with the command "make px4-v2" in the window 7：
"mavgen: success"; exit 0; else echo "mavgen: failed"; exit 1; fi
/bin/sh: python: command not found
mavgen: failed
make: *** [/d/pixhawk_toolchain/ardupilot/Build.ArduCopter/libraries/GCS_MAVLink
/include/mavlink/v2.0/ardupilotmega/mavlink.h] Error 1
Please help me with this issue.

#### Version
today's git clone

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad X

#### Hardware type
pixhawk

magicrub, 2017-03-21:
It's missing Python. Do you have that installed? Is it in your path?

QiMo2016, 2017-03-21:
@magicrub The pixhwak_toolchain v16 has included Python27 and pointed to the path:
export PATH=/bin:/usr/bin:/usr/local/bin:/c/windows/system32:d:/pixhawk_toolchain/Python27:d:/pixhawk_toolchain/Python27/Scripts:d:/pixhawk_toolchain/toolchain/bin:$PATH

OXINARF, 2017-03-21:
This should belong in the forum or in Gitter, not here.

How are you opening the command-line you are using to compile? Have you tried a `make px4-clean` and then building again?

OXINARF, 2017-04-25:
After more than a month there was no answer. Since this doesn't belong here anyway, I'm closing it.

closed 2017-04-25 by OXINARF