ArduPilot/ardupilot #5923: Compass can assume wrong offsets

mboland, 2017-03-21:
#### Issue details
When a compass is calibrated it's offsets are set and relate as ID#. e.g.:
DEV_ID #1 = OFS_X/Y/Z #1  DEV_ID2 = OFS2_X/Y/Z
with any external compass taking ID#1, then ID#2, followed by internal compasses.

However

If an external compass in removed, or has failed to be seen at boot time for whatever reason, the internal compass designation is changed and brought forward in ID#, but the offsets are not.

So, if I have a Pixhawk with one external compass, both calibrated and flying, and the external is not seen at boot, then the internal compass becomes ID#1 but will be using the offsets for the external compass.

Also

If an external compass is deliberately removed, the ID for the internal is used for #1 but also remains for ID2, and because there is already offsets in place may not ask for recalibration.
The duplication of internal ID in position #1 and #2 persists between boots.

This came about chasing a fault that turned out to be a dodgy (technical term) I2C breakout giving intermittent connections. The effects in flight can be dramatic.

#### Version
3.5rc2 but most likely prior versions as well.

#### Platform
[ X] All?
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Octo

#### Hardware type
Pixhawk1 and Pixhawk2.1

#### Logs
Pixhawk2.1 w/2 external compass

[image]
[image]

Now with the external compass removed

[image]
[image]

OXINARF, 2017-03-21:
> If an external compass in removed, or has failed to be seen at boot time for whatever reason, the internal compass designation is changed and brought forward in ID#, but the offsets are not.

As expected - but you are supposed to need to do a new compass calibration. Do you have your pre-arm checks enabled?

> So, if I have a Pixhawk with one external compass, both calibrated and flying, and the external is not seen at boot, then the internal compass becomes ID1 but will be using the offsets for the external compass.

As I said, you are supposed to not be able to arm without redoing compass calibration.

> If an external compass is deliberately removed, the ID for the internal is used for ID1 but also remains for ID2, and because there is already offsets in place may not ask for recalibration or indicate any error.
The duplication of internal ID in position 1 and 2 persists between boots.

That shouldn't be a problem since it means no compass was found in ID2.

mboland, 2017-03-21:
> As expected - but you are supposed to need to do a new compass calibration. Do you have your pre-arm checks enabled?

Full pre-arm checks are ON.

It seems that if the offsets are good for both then no compass error is generated. So in fact it cannot be known at takeoff which compass is in use.

So what you are saying is that there is a compass check for a change from the last boot.
It is not working.

OXINARF, 2017-03-21:
> So what you are saying is that there is a compass check for a change from the last boot.
It is not working.

It should check for the generated compass id and compare it with the saved one, if they are different, you need to redo the calibration.

Do you have a log (and maybe a tlog) you can share with this issue?

Thanks for the report Mike.

mboland, 2017-03-22:
If I can't find the last flight logs I will generate some new ones as soon (if) it stops raining here.

mboland, 2017-03-22:
The last flight was interesting to say the least, but it did hi light the compass connection problem for me as being physical, the I2C breakout board. [Logs here](http://uavs.com.au/public_html/Arducopter/logs/2017-03-17/2017-03-17%2007-35-53.log.zip)
The result of the compass error was a 30deg track variance to the left for all stick inputs in Loiter.
I will retest a soon as the water stops falling (whenever that might be, >250mm last 4 days and counting.

rmackay9, 2017-03-25:
I tested this on my EnRoute EX700 with a pixhawk1.. I connected an external GPS/compass and it complained I needed to do the compass calibration.  I did the calibration, rebooted and then disconnected the GPS/compass and again it complained it needed a compass calibration.

So I tend to think the pre-arm check code is working correctly.

closed 2017-03-25 by mboland

OXINARF, 2017-03-25:
@mboland Have you got it working?

mboland, 2017-03-25:
No, but if it's been tested and verified as working then it must be something in my configuration that is causing it, so I need to go back to the bench and start testing again.

OXINARF, 2017-03-25:
Well it worked for Randy, but doesn't mean there isn't a bug hidden somewhere. I haven't had a chance to look at your logs, but please test again and if it still doesn't work for you then re-open this - if a bug exists here it would be a very important one.