ArduPilot/ardupilot #5925: Copter: Airmode feature request

Gunaveh, 2017-03-22:
https://oscarliang.com/betaflight-airmode/

Please provide this type of function in a future release.  I use arducopter on my racing quads and when dealing with agressive descents i often encounter a large degree of instability in pitch roll and yaw at low throttle settings.  I would love to have acro stability even when bumping the throttle stick off its lower limit.

Pedals2Paddles, 2017-03-23:
Funny should mention this. We were just looking at crash logs from a Solo user on the Solo forum that involved this situation.  Operator was descending rapidly down a mountainside in stabilize mode. 40+mph, 100ft per second descent rate, pitched forward 35 degrees. So he had the throttle pretty low to make like a brick and descend this quickly.  But along the way his thumb inadvertently brought the throttle to minimum.  The motors obviously cut back to minimum armed with no stabilization. And such was the end of that adventure.

If we're in stabilize, shouldn't it stay "active" unless you have the stick full down **and fully left?**  Is there a reason it goes to idle even in flight?

lvale, 2017-03-23:
Just adjust the MOT_SPIN_MIN to a value you feel confortable 

http://ardupilot.org/copter/docs/parameters.html#mot-spin-min-motor-spin-minimum

Pedals2Paddles, 2017-03-23:
That doesn't do it. That just adjusts the speed. It cease stabilizing too. All motors reduce to minimum armed, period. No pitch, roll, or yaw is controlled. That's the issue/request here.

lvale, 2017-03-23:
Just tested on SITL and still have PRY control.
Quick test for you. Get a drone high on SITL on Guided Mode. Switch to Circle mode and TH at 0, The drone comes falling down on a corkscrew ie maintaning the PRY attitude and falling as it should

rmackay9, 2017-03-23:
I think this is a valid request.  In manual modes (Stabilize, Acro) we do turn off stabilization if the throttle is brought to zero.  The danger though is when the pilot wants to land in stablize or acro mode.  The bump on the ground could lead to the vehicle not recognising that it's landed and this could make it impossible for the user to disarm the vehicle.

A solution would be to keep stabilizing when the throttle is zero only if the user also has an auxiliary switch setup to turn off the motors (aka interlock, motor e-stop, kill switch).  If such a switch is setup then the user could easily use it if he/she has problems disarming in the normal way.

Gunaveh, 2017-03-23:
That is an issue in betaflight as well. Landing is not suggested in airmode and it is made a seperate flight mode from acro.
 I always land my race quads using the e stop  so i personally wouldnt have a problem with rmackay's implementation.

Gunaveh, 2017-03-23:
Ivale. I believe this is only an issue in manual throttle modes. Stabilize, acro.

Pedals2Paddles, 2017-03-23:
If down and left is the disarm command, why would you need a motor interlock to stop the motors? Full down should bring the motors to minimum but maintain stabilization at all times, ignoring landing detection. If you're landing, *down and left* disarm regardless of what landing detection thinks. No motor interlock needed. This is how I always remember stabilize working from a long time ago (pre 3.3 anyway). Was this how it used to be, or am I just nuts? Now it seems to be the opposite of this.

geofrancis, 2017-03-23:
you could set up an idle up switch on your handset so that it keeps the throttle from going down to the point of idling.

peterbarker, 2017-04-02:
On Thu, 23 Mar 2017, Matt wrote:

> If down and left is the disarm command, why would you need a motor interlock
> to stop the motors? Full down should bring the motors to minimum but
> maintain stabilization at all times, ignoring landing detection. If you're

Thoroughly disagree.  STAB mode should remain as raw as possible as it is 
the primary recovery mode for Copter.  We should never stop the user from 
crashing the vehicle in STAB, if that's what she wants to do.  And I think 
ACRO goes without saying.

Adding in more modes and/or switch-based modifiers seems reasonable to me, 
'though.

Saxin, 2017-05-15:
Guys just experienced exactly that today:

1. Stab mode - sometimes one want to descent quickly yet remain stabilized, it's seems attitude authority is lacking in such case, in my case the copter is locked in weird (around 30 deg to horizon) angle while free falling.

2. Acro mode - basically the same issue one can descent rapidly yet want to perform a quick maneuver during this free fall say roll for the sake of example, it can help if full motor authority is available for the move / maneuver.

I've tried increasing the MOT_SPIN_MIN even to 0.18 with no apparent affect.

peterbarker, 2017-05-16:
On Mon, 15 May 2017, Eddi Maevski wrote:

>     Stab mode - sometimes one want to descent quickly yet remain stabilized,
>     it's seems attitude authority is lacking in such case, in my case the
>     copter is locked in weird (around 30 deg to horizon) angle while free
>     falling.

zero-throttle in stabilize should remain no-significant-motor-output.

Any reason you can't use ALTHOLD?  It will retain attitude as you're 
indicating descent rate not throttle.

Gunaveh, 2017-05-16:
Thats fine, but could we have full authority in acro mode while at zero throttle?  Ive had to switch to betaflight as arducopter is just not as capable for fpv racing.

Saxin, 2017-05-16:
@peterbarker - I'm not proposing to change that for default, proposal is for special add on or "modifier" mode, initiated by a switch to allow full authority... 
This issue is raised for ACRO mode, which sometimes dictates to agile flight, controlled modes provide predetermined descent rate - but we have another discussion here I think.

@Gunaveh - I'll try to double check in Acro but if I remember correctly I got more or less same tumble (change in the requested angle) during free fall, per my understanding this should not happen - controller should have full authority even at min throttle.

OXINARF, 2017-05-18:
Dev team member @R-Lefebvre, developer of the motor interlock feature, as brought to my attention that when using interlock the system continues to try to stabilize even when putting throttle to zero. So this is something you can try - it wasn't designed for it though, so no promises that it absolutely works.

lthall, 2018-01-11:
Hi all,
This was specifically what motor interlock feature does. I use this all the time to ensure the aircraft is stabilized at all times. There is also a parameter ATC_MOT_MIN_MAN that defines how much the average motor output can be increased to stabilize the aircraft at minimum stabilizing throttle.

So you can use the interlock and then adjust ATC_MOT_MIN_MAN to your taste.

closed 2018-01-11 by lthall