ArduPilot/ardupilot #5932: XY Position moved instead of "NED" for no_gps

Ahrovan, 2017-03-24:
When we set ekf_gps_type to no_gps, use Optical flow, the best situation for see position moved from take-off (zero pos) to target position is simple coordinates instead of "NED"...
NED position is North East down coordinates that is better for GPS targets

rmackay9, 2017-03-25:
@Ahrovan, could you clarify a bit more?

I guess you're using the Copter firmware... which flight mode are you using?  I guess you're using Guided mode and perhaps dronekit to send movement comments?  .. or are you clicking on a map?  what ground station are you using?

Ahrovan, 2017-03-25:
Yes. Arducopter 3.5 rc2. Loiter flight mode with no_gps (optical flow)
I am useing mission planner .. but for optical flow (indoor) movment clicking on the map and NED coordinates is meaningless (?) But i think change NED to xy coordinates is usefull

Ahrovan, 2017-03-26:
@rmackay9 
I tested PE,PN,PD values, but with optic flow, error is Greater that 0.5m
[image]
[image]

rmackay9, 2017-03-27:
So what you're trying to do is use optical flow in Loiter mode using AC3.5.

You've gone through the calibration described on this page?
http://ardupilot.org/copter/docs/common-px4flow-overview.html

Optical flow reports the vehicle's velocity and then we integrate that to get a position so the position drifts.  The amount it drifts can depend upon the lighting and the surface it's looking at.  Also the range finder needs to be accurate as well.

The graph you've posted doesn't look too bad.  It looks like at the end of the test it's within about 1m of it's original position.  That's not bad I think.

Reports on beta software should be posted to our discourse server here: http://discuss.ardupilot.org/c/arducopter/copter-3-5... so I will close this issue but feel free to raise a new discussion over there.

closed 2017-03-27 by rmackay9