ArduPilot/ardupilot #5951: AP_Airspeed: Failed sensor data propagates through all control loops

WickedShell, 2017-03-29:
#### Issue details
First comment: I hope to be proven wrong on this.

We can have a failed sensor for a variety of reasons, in this case we are considering a failed sensor to be anytime AP_Airspeed::healthy() would return false (no valid data from the sensor within the last 0.1 seconds). We use this when we emit the airspeed healthy flag to the system. However that doesn't stop us from using the data after that.

The first failure is that when we compute the raw airspeed value (this is what you see logged for airspeed data) we ask AP_Airspeed::get_pressure() for a pressure value. If we don't succeed in getting a differential pressure we mark ourselves as not healthy. However we return 0, which since the sensor health isn't checked within read() allows the value to propagate through the system. We then add our offsets, multiply by the ratio and take the square root for the _raw_airspeed value. This will always result in a fixed value. In my example an offset of around 106 and a ratio of 1.9 leads to a fixed airspeed of 14.1 being reported).

We then integrate this data into the _airspeed estimate. The problem is that if we take a library like TECS it simply calls AP_AHRS::airspeed_estimate() which just checks that the sensor is enabled. It then fetches the current airspeed (which will be the 14.1 value after it makes it through the filter) and uses it. This means that a failed sensor will perpetually be read at this value, and your control loops will all be based around a fixed estimate, and not allow any of the fallback code to run (such as TECS has some code that attempts to manage this scenario).

I'd **strongly** like to be wrong about this, so please double check my audit of the code involved with this. If you have any log data that returns 0.0 as the airspeed temperature, and you had an airspeed sensor enabled you would have been affected by this.

#### Version
Master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

magicrub, 2017-03-29:
I have a log of my pitch flipping out when intentionally applying an abnormal ratio. That didn't change the healthy() flag but I know serious abnormal behavior can propagate..

WickedShell, 2017-03-29:
airspeed_sensor_enabled() calls healthy() and I missed that. Excellent.

closed 2017-03-29 by WickedShell