ArduPilot/ardupilot #5952: DataFlash inconsistency between declared and actual msg length

Georacer, 2017-03-29:
#### Issue details
We discovered with @hunt0r that in the latest firmware version, the length of the BAR2 and BAR3 messages is declared to be 37 bytes long in the corresponding FMT messages.
But actually, the format string 'QffcfIf' corresponds to 30+3 bytes.

Actually, older versions of ArduPlane had similar issues, but currently these are they only affected message types.

#### Version
V3.8.0beta3 (21951569)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Any

#### Hardware type
N/A

#### Logs
http://autotest.ardupilot.org/ArduPlane-log.bin

WickedShell, 2017-03-29:
This was my bad, introduced with #5930 by accident. I updated the BARO message, and forgot to get BAR2 BAR3 (They aren't ordered next to BARO the way GPS ones are aligned and I had forgotten about the second/third instances at the time)

OXINARF, 2017-03-29:
Thanks for the report, fixed with https://github.com/ArduPilot/ardupilot/commit/ec5bbbd2d026c8ebead9cc5e41b8679023a89ac4. This was only two days old, good catch!

closed 2017-03-29 by OXINARF