ArduPilot/ardupilot #5955: Statustext severity improvements

AndKe, 2017-03-30:
#### Issue details
telemetry statustext is not defining severity as good as it could.

#### Version
AP 3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover

```
2017-03-22 11:28:17.61: STATUSTEXT {severity : 6, text : Flight plan received}
2017-03-22 11:37:56.32: STATUSTEXT {severity : 6, text : Throttle armed}
2017-03-22 11:38:49.36: STATUSTEXT {severity : 6, text : Executing nav command ID #22}
2017-03-22 11:39:02.21: STATUSTEXT {severity : 6, text : Armed AUTO, xaccel = 13.1 m/s/s, waiting 0.0 sec}
2017-03-22 11:39:02.26: STATUSTEXT {severity : 6, text : Triggered AUTO. GPS speed = 1.9}
2017-03-22 11:39:04.31: STATUSTEXT {severity : 6, text : Holding course 25412 at 11.3m/s (-22.6)}
2017-03-22 11:39:24.95: STATUSTEXT {severity : 6, text : Takeoff level-off starting at 4m}
2017-03-22 11:39:26.55: STATUSTEXT {severity : 6, text : Takeoff complete at 60.11m}
2017-03-22 11:39:26.58: STATUSTEXT {severity : 6, text : Executing nav command ID #16}
2017-03-22 11:40:09.26: STATUSTEXT {severity : 6, text : Reached waypoint #2 dist 4m}
2017-03-22 11:41:09.24: STATUSTEXT {severity : 4, text : EKF2 IMU0 switching to compass 1}
2017-03-22 11:41:09.49: STATUSTEXT {severity : 4, text : EKF2 IMU1 switching to compass 1}
2017-03-22 11:43:04.74: STATUSTEXT {severity : 6, text : Executing command ID #178}
2017-03-22 11:43:04.75: STATUSTEXT {severity : 6, text : Set airspeed 15 m/s}

```
important / dangerous things like throttle arming is logged at the same level as waypoint.    same for critical parameters like airspeed.
By fine-adjusting some levels, we could easily better distinguish between messages that needed attention/voice/alerts , and those trivial info messages. 

I'd say it would be good idea to make better use of 4=warning , 5=notice 6=info this way:

```
2017-03-22 11:28:17.61: STATUSTEXT {severity : 6, text : Flight plan received}
2017-03-22 11:37:56.32: STATUSTEXT {severity : 4, text : Throttle armed}
2017-03-22 11:38:49.36: STATUSTEXT {severity : 6, text : Executing nav command ID #22}
2017-03-22 11:39:02.21: STATUSTEXT {severity : 5, text : Armed AUTO, xaccel = 13.1 m/s/s, waiting 0.0 sec}
2017-03-22 11:39:02.26: STATUSTEXT {severity : 6, text : Triggered AUTO. GPS speed = 1.9}
2017-03-22 11:39:04.31: STATUSTEXT {severity : 6, text : Holding course 25412 at 11.3m/s (-22.6)}
2017-03-22 11:39:24.95: STATUSTEXT {severity : 6, text : Takeoff level-off starting at 4m}
2017-03-22 11:39:26.55: STATUSTEXT {severity : 6, text : Takeoff complete at 60.11m}
2017-03-22 11:39:26.58: STATUSTEXT {severity : 6, text : Executing nav command ID #16}
2017-03-22 11:40:09.26: STATUSTEXT {severity : 6, text : Reached waypoint #2 dist 4m}
2017-03-22 11:41:09.24: STATUSTEXT {severity : 4, text : EKF2 IMU0 switching to compass 1}
2017-03-22 11:41:09.49: STATUSTEXT {severity : 4, text : EKF2 IMU1 switching to compass 1}
2017-03-22 11:43:04.74: STATUSTEXT {severity : 6, text : Executing command ID #178}
2017-03-22 11:43:04.75: STATUSTEXT {severity : 5, text : Set airspeed 15 m/s}

```

OXINARF, 2017-03-30:
Severity is highly subjective, not everyone will agree with the level for all messages.

I think a PR with these changes would be much more useful for evaluation by the Plane maintainers, but since you have detailed what you think should be the levels for certain messages I'll leave this open.

AndKe, 2017-03-30:
I just probed the possibility for getting such PR accepted, I will go thru all AP STATUSTEXTs  and make my suggestions.

magicrub, 2017-03-30:
It's no subjective, the priority definiteions are listed somewhere... I'm having trouble finding where it is though

WickedShell, 2017-03-30:
@magicrub It's the MAV_SEVERITY enum http://mavlink.org/messages/common

```
0	MAV_SEVERITY_EMERGENCY	System is unusable. This is a "panic" condition.

1	MAV_SEVERITY_ALERT	Action should be taken immediately. Indicates error in non-critical systems.

2	MAV_SEVERITY_CRITICAL	Action must be taken immediately. Indicates failure in a primary system.

3	MAV_SEVERITY_ERROR	Indicates an error in secondary/redundant systems.

4	MAV_SEVERITY_WARNING	Indicates about a possible future error if this is not resolved within a given timeframe. Example would be a low battery warning.

5	MAV_SEVERITY_NOTICE	An unusual event has occured, though not an error condition. This should be investigated for the root cause.

6	MAV_SEVERITY_INFO	Normal operational messages. Useful for logging. No action is required for these messages.

7	MAV_SEVERITY_DEBUG	Useful non-operational messages that can assist in debugging. These should not occur during normal operation.
```

A PR would be the best approach, we can definitely rework some of them, but without seeing the specific ones in that context it's hard to evaluate.

AndKe, 2017-03-30:
Yes, I need to think a bit more about it , the thing is that I would like to easily filer trivial items as 6(INFO) from speech /visual warnings, but at the same time, there's a wide use of INFO, even for messages I do not agree should be "info" only, like obvious error messages:
```

quadplane.cpp:445:        GCS_MAVLINK::send_statustext_all(MAV_SEVERITY_INFO, "Not enough memory for quadplane");
quadplane.cpp:567:    GCS_MAVLINK::send_statustext_all(MAV_SEVERITY_INFO, "QuadPlane setup failed");
quadplane.cpp:578:            GCS_MAVLINK::send_statustext_all(MAV_SEVERITY_INFO, "QuadPlane setup failure for %s",
motor_test.cpp:81:        plane.gcs_send_text(MAV_SEVERITY_INFO, "Must be disarmed for motor test");
```

All I really am trying to achieve, is to have purely non-exciting messages like "reached waypoint xx" excluded,  while not excluding "MAV_SEVERITY_INFO,"Land final started");"   

so .. maybe we could change 7 to be "trivial" , and level 8 debug ?   - a change in mavlink would be needed..

Of course, another solution would be to filter at GCS end.. just looking for text.

WickedShell, 2017-03-30:
You unfortunately can't change the MAVLink definition without breaking quite a lot of existing implementations. You can just tack a new field onto the enum however.

Honestly I'm personally of the opinion that the "reached waypoint xx" messages could be entirely dropped, I don't care about seeing them, and a GCS could easily create the message for you if you felt strongly about it. However I was also the minority opinion on this last time it came up, so I don't expect a lot of success from that type of change...

Those error messages you just linked though I'd agree definitely look like very strong candidates for MAV_SEVERITY_ERROR and we can do that change quite reasonably.

OXINARF, 2017-03-30:
@magicrub @WickedShell I'm aware of that enum, but that's not enough, what one person thinks fits in a level, another has a different opinion. If it wasn't subjective we would always get it right and no discussion was needed.

@AndKe I think there's enough levels already. I believe the bigger issue is in what messages the GCS show rather than the classification we do. MP for example only shows in the HUD beginning at a certain level (warning I think) but has several whitelist inclusions - meaning some messages will be shown independently of their level.

WickedShell, 2017-03-30:
@OXINARF well for example all of the ones @AndKe just linked in quadplane/motor_test I think are clearly errors, and that can be improved regardless.

OXINARF, 2017-03-30:
@WickedShell I agree they aren't info (but someone thought that was adequate), but why error and not critical or even emergency? This is what I'm saying, it's not black and white, several levels can apply to a message depending on a personal view.

lvale, 2017-04-02:
link to this https://github.com/ArduPilot/ardupilot/issues/3772

peterbarker, 2021-07-30:
I think we need a PR with concrete changes for this issue.

If someone feels like putting together a concrete proposal in a PR, please do so.

closed 2021-07-30 by peterbarker