ArduPilot/ardupilot #5958: pwm value exceed 2000(rover)

oceangis, 2017-03-30:
#### Issue details
i use rover 3.1.2 and send pwm value though mavlink by joystick that pwm value was calibrated to 1000-2000. when i push to the maximum throttle,the pwm value of channel 1 and 3 displayed is about 2250.

#### Version
rover 3.1.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ × ] Rover

#### Hardware type
pixhawk

khancyr, 2017-03-30:
Hello, Yes we are aware of this bug, we are working on a fix : see https://github.com/ArduPilot/ardupilot/pull/5819 (If you are only using rover, it should be safe to appy those fix while we are merging them into master)

oceangis, 2017-03-30:
Hi, khancyr 
    Thank you very much.

OXINARF, 2017-03-30:
This is a duplicate of #5817, closing.

closed 2017-03-30 by OXINARF