ArduPilot/ardupilot #5959: AC_PrecLand EKF does not honour LANDING_TARGET time_usec

fnoop, 2017-03-30:
#### Issue details
AC_PrecLand EKF does not take LANDING_TARGET time_usec into account.  Therefore it processes and applies IMU correction at the wrong time and incorrect (and potentially dangerous) attitude adjustments are produced.

#### Version
3.4, 3.5, current master

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

ClementChristomanos, 2017-04-19:
@fnoop Did you or someone get a chance to take a look at this ? It looks like the current EKF approach could be buggy but in the long term it could lead to higher performance.

fnoop, 2017-04-19:
@ClementChristomanos No @jschall said something on gitter recently that he was contemplating rewriting the precland ekf, so I didn't spend any time on it.   Precland is effectively broken, unless using low-latency input or irlock, there probably should be a red warning up somewhere.

ClementChristomanos, 2017-04-20:
@fnoop Thanks for that info :)

patrickpoirier51, 2017-06-26:
Here is a simple python script that clearly show the behaviour of the "mini EKF" within SITL.
Test procedure:
- You need Dronekit Python Installes
-  Install the sonar rangefinder as described in wiki: http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#adding-a-virtual-rangefinder
- Set these parameters  
PLND_ENABLED     1
PLND_EST_TYPE    0
PLND_TYPE        1
CH7_OPT          39

Launch the python script on the same Linux system (It will publish @ 0.0.0.0:5763)
Start SITL
mode loiter
Arm Throttle
rc 3 1800
get to 20 meter
rc 3 1500 (stabilize altitude)
rc 7 1900 = activate precision loiter

The quad should drift just a little bit.
Now type:   param set PLND_EST_TYPE  1
You should see the quad starting the "Psycho Dance"
To get it even worse, go in the python script and change x = 0 to x = 1
Relaunch script

If you switch back to RAW data (no estimator) it does stabilize
param set PLND_EST_TYPE  0

Hope this will help to investigate

[Precision_Landing_Dummy.py.zip](https://github.com/ArduPilot/ardupilot/files/1101905/Precision_Landing_Dummy.py.zip)

MohamedAliRashad, 2019-04-08:
Is there any updates on this as i would like to tackle it in my GSoC program this year ?
And i will appreciate it a lot if someone can guide for more resources about PrecLand and EKF ^ ^