ArduPilot/ardupilot #5960: Copter 3.5 messaging says v2 even when v3 is loaded

Pedals2Paddles, 2017-03-30:
#### Issue details
On at least 3.5rc-2 and 3.5rc-3, the mavlink messages contradict the version loaded. I've loaded the v3.px4 firmware for both 3.5rc-2 and 3.5rc-3.  In both of them, the messaging says v2 even though I loaded v3.  So the question here is... is this just a typo in the firmware file and it really is v3?  Or is this thing getting compiled wrong and the v3 file is really v2?

[img]https://i.imgur.com/5vteBQC.jpg[/img]

#### Version
Noticed on Copter 3.5rc-2 and Copter 3.5rc-3. 

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
Pixhawk 2 from a 3DR solo

#### Logs
This is a dataflash log from a flight on this firmware.
https://drive.google.com/open?id=0BwxNJyHTG9XJSjljVWNCZjFybjg

OXINARF, 2017-03-30:
We can change this, but it is expected that v3 shows as v2.

Pedals2Paddles, 2017-03-30:
OK cool. I mainly wanted to verify I indeed had the right firmware on this thing.  Thanks!

tridge, 2017-03-31:
it is the current expected behaviour, but I agree it should be changed

OXINARF, 2017-04-03:
Fixed with #5982. Thanks for the report!

closed 2017-04-03 by OXINARF