ArduPilot/ardupilot #5975: Dataflash log does not contain altitude type in logged mission items.

AndKe, 2017-04-01:
dataflash message type CMD , looks like
CMD, Format : QHHHfffffff, Columns : TimeUS,CTot,CNum,CId,Prm1,Prm2,Prm3,Prm4,Lat,Lng,Alt}

There is no indication of relative or absolute (or other) altitude for any loged waypoint/mission item. 
The altitude-type should be stored too.

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x] Plane
[  ] Rover

jmachuca77, 2017-04-03:
I am not sure how this is handled, for the waypoints there is a coordinate frame that has to be sent somehow, that is what tells the autopilot if the waypoint alt is relative or absolute. But this seems not to be included in the dataflash log. I checked mavmission.py which extracts the mission from the dataflash log and it has the frame field value for relative alt hardcoded into it.

enjoybeta, 2018-03-25:
#7936 just noticed this pull

WickedShell, 2018-03-25:
Sorry, hadn't even seen this was an issue when I sent the other PR in. Anyways fixed in master and next release.

closed 2018-03-25 by WickedShell