ArduPilot/ardupilot #5979: best way to build the copter code, and SITL

engineerLU, 2017-04-02:
#### Issue details
hello,
have a ice day, I am a control engineer and I am working on a master thesis concerning fault tolerance for a hexarotor. I am using pixhawk hardware.  
I want to  ask about the best way to build the copter code, and SITL for PX4 hardware. My OS is Windows 10. I have tried ubuntu bash and there was a lot of problems?
I am thinking to use "Building for Pixhawk/PX4 using Eclipse on Windows" and "Setting up SITL on Windows" as mentioned in ardupilot.org. 
I also wonder if the make build is still working well?
Thanks a lot.
#### Version
_What version was the issue encountered with_

#### Platform
Copter
#### Airframe type
hexa

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

Nishchay-sopho, 2017-04-03:
The website [http://ardupilot.org/dev/docs/sitl-native-on-windows.html] provides a very description about setting up sitl on a windows pc. I have used it to setup sitl on my pc and it works fine.

OXINARF, 2017-04-03:
@engineerLU This issues list is just for confirmed bugs or feature requests. For developer support questions either post in the forum (http://discuss.ardupilot.org) or in Gitter (http://gitter.im/ArduPilot/ardupilot)

As @Nishchay-sopho pointed out we have a guide for Windows on our docs. You can also use the "Bash on Ubuntu on Windows", but we don't have a specific guide for SITL for that environment - the one for Linux should work quite well though: http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html

closed 2017-04-03 by OXINARF